calyxir/calyx #695: [fud] Usage of "kernel" naming convention in the HLS flow

mortbopet, 2021-09-27:
In general, hardcoded top-level function/filenames are in general expected to be called `main` in various `.tcl` scripts.
However, for the Vivado HLS flow, top-level functions and files are named `kernel`. This creates an inconvenience for flows which start at e.g. C level that may go through either the Vivado HLS flow, or (via Polygeist) through the Calyx flow, in that there has to be a manual intervention somewhere in there to rename the top-level function from "kernel" to "main".

A proposed fix is to rename "kernel" to "main" in https://github.com/cucapra/calyx/blob/master/fud/synth/hls.tcl and https://github.com/cucapra/calyx/blob/master/fud/fud/stages/vivado/stage.py#L141.

rachitnigam, 2021-09-27:
Another possible option is providing a command-line parameter to the tcl script to specify the name of the top-level module. This can then be changed using `fud`'s `-s <stage> ...` stuff during a command invocation. How hard would it be to do that?

mortbopet, 2021-09-27:
That would definitely also solve the problem. That, or a global parameter denoting the top-level function, which will be used anywhere where we currently use `main` or `kernel`.

rachitnigam, 2021-09-27:
Cool. If you'd like, open a PR. I'm currently working on some other things so it'll be a while before I can get to this.

sampsyo, 2021-09-28:
Seems like a good idea to me!

PR #1773 (2023-11-13): `fud` stages for HLS place and route
Adds stages for running Vivado HLS files through place and route. This should probably also close #695, since it adds an option to change the name of the top-level function.
closed 2023-11-13 by bcarlet