calyxir/calyx #699: [fud] Missing minimum version for networkx

ohxh, 2021-09-28:
I was working through the [getting started guide](https://capra.cs.cornell.edu/docs/calyx/) and ran into the following error: `fud` seems to depend on the python package `networkx`, specifically the method `all_simple_edge_paths`. It's hard to tell from the changelog but it seems the method was introduced sometime between 2.4 (my version) and 2.6 (current).

```
calyx % fud e examples/tutorial/language-tutorial-iterate.futil \
  -s verilog.data examples/tutorial/data.json \
  --to dat -v
```
Produces the following output.
```
Traceback (most recent call last):
  File "/usr/local/anaconda3/bin/fud", line 8, in <module>
    sys.exit(main())
  File "/usr/local/anaconda3/lib/python3.7/site-packages/fud/main.py", line 295, in main
    exec.run_fud(args, cfg)
  File "/usr/local/anaconda3/lib/python3.7/site-packages/fud/exec.py", line 67, in run_fud
    path = construct_path(args, config, args.through)
  File "/usr/local/anaconda3/lib/python3.7/site-packages/fud/exec.py", line 45, in construct_path
    path = config.REGISTRY.make_path(source, target, through)
  File "/usr/local/anaconda3/lib/python3.7/site-packages/fud/registry.py", line 46, in make_path
    all_paths = list(nx.all_simple_edge_paths(self.graph, start, dest))
AttributeError: module 'networkx' has no attribute 'all_simple_edge_paths'
```

This can be fixed as needed by manually upgrading (`pip install networkx --upgrade`), but adding an explicit dependency on `"networkx^2.6"` should solve this for any future installs with old versions of networkx present.

cgyurgyik, 2021-09-28:
Thanks for pointing this out!

rachitnigam, 2021-09-28:
Hm, there is a dependency on the `networkx` package and I assumed it'd install the right one: https://github.com/cucapra/calyx/blob/master/fud/pyproject.toml#L21

@ohxh do you want to take a shot at fixing the issue and opening a PR?

sampsyo, 2021-09-28:
Indeed; a version constraint here would be awesome, if you don't mind throwing together a quick PR, @ohxh:
https://github.com/cucapra/calyx/blob/84e906e523569a75800e40ec15661ee759593c4d/fud/pyproject.toml#L21

(FWIW, the way `pip` works is that it doesn't install anything if the dependency is already satisfied on the system—as it would be if you happen to have manually, separately installed the same package. So we should declare the minimum version we require.)

ohxh, 2021-09-29:
@rachitnigam @sampsyo I'm just getting involved in this project and can't branch / PR in the repo. I'm talking with @EclecticGriffin on Thursday and I'll revisit after that. Probably the rest of these deps should get minimum versions too

sampsyo, 2021-09-29:
Cool! FWIW, it's also totally cool to submit a PR from your own fork (as opposed to a branch in this repo):
https://docs.github.com/en/github/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

commit 45dbdb52 (2021-10-05): Specify minimum version for networkx. Fixes #699
closed 2021-10-05 by rachitnigam