calyxir/calyx #702: [CIRCT] Groups not emitting the attributes.

cgyurgyik, 2021-10-02:
From [tests/backend/mlir/attributes.futil](https://github.com/cucapra/calyx/blob/master/tests/backend/mlir/attributes.futil):
```
// -p well-formed -b mlir
import "primitives/core.futil";
import "primitives/binary_operators.futil";
component main<"static"=1>(@stable(32) in: 32, @go go: 1, @clk clk: 1, @reset reset: 1) -> (@stable(0) out: 32, @done done: 1) {
  cells {
    @precious r = std_reg(32);
    @external(32) le = std_le(32);
  }
  wires {
    group upd<"stable"=1> { // <-- No `stable` attribute added to `upd`.
      upd[done] = r.done;
    }
    comb group cond<"stable"=0> {
    }
  }
  control {
    @bound(32) while le.out with cond {
      @static(2) upd;
    }
  }
}
```

From the `.expected`:

https://github.com/cucapra/calyx/blob/84e906e523569a75800e40ec15661ee759593c4d/tests/backend/mlir/attributes.expect#L7-L9

rachitnigam, 2021-10-02:
@cgyurgyik What does the expected output look like?

cgyurgyik, 2021-10-02:
Similar to other containers, where the attribute dictionary is after the
body, e.g.

‘’’
Calyx.group @g { … } { stable=1 }
‘’’

(I might be off a bit somewhere there)

cgyurgyik, 2021-10-02:
(new comment since E-mail replies don't support MD):

```
calyx.group @upd {
      calyx.group_done %r.done : i1
    } {stable=1}
```
```

cgyurgyik, 2021-10-02:
Taking care of this.

PR #706 (2021-10-03): [CIRCT] Add attributes for groups.
Closes #702.
closed 2021-10-03 by cgyurgyik