calyxir/calyx #703: [CIRCT] Need to add entry component name to `calyx.program`.

cgyurgyik, 2021-10-02:
`calyx.program` now expects the name of the entry component. Examples found [here](https://github.com/llvm/circt/blob/main/test/Dialect/Calyx/compile-control.mlir).

Concretely, this is just:
```
calyx.program {          // Before
calyx.program "main" {   // After

cgyurgyik, 2021-10-04:
Any thoughts on this one @rachitnigam ? 

When we go from MLIR --> native, we lose this information. Some options:

0. Add some comment when emitting that remarks on which component is the entry-point. Gross.
1. We *could* add some key value pair to FUD, e.g. `-entrypoint "main"`.
2. Is there any benefit to marking the entry-point component in Calyx?

`component main<"entrypoint"=1>() -> () { ... }`

rachitnigam, 2021-10-05:
#708 mentions that we need a `toplevel` annotation on components to work with simulation. Maybe we can reuse that annotation? If there isn't any component with that, we can default to `main`. How does that sound?

cgyurgyik, 2021-10-05:
I think fixing this with `fud` is fine, but do you foresee this as a re-occurring issue? 

Specifically, the frontend to Calyx defines a specific component as the entry-point, and the backend needs to determine which component is the entry-point. To me, this should be information kept at the program level rather than the driver level.