calyxir/calyx #731: Tracking overflow / underflow in mathematical operations.

cgyurgyik, 2021-10-17:
I think this would be great for debugging / program correctness purposes.

I'm unsure how easily this could be added to the interpreter. @EclecticGriffin, perhaps you have an opinion here?

Another option I foresee is tackling this at the primitive / component level. For each component / primitive add an `{over, under}flow` flag. This also requires adding the additional logic to check for this case. The benefit here is allowing the user to handle this case directly as well.

EclecticGriffin, 2021-10-17:
This isn't particularly bad, the interpreter currently under/over flows as appropriate though it is relatively simple to escalate to an error if desired.

sampsyo, 2021-10-18:
This seems super cool—a perhaps "easy" win for one of the interpreter's main superpowers, which is catching and reporting errors that would be expensive or impractical to enforce in the real generated hardware. Perhaps a sensible way to go about it would be for this to be a warning by default and promoted to an error as an option.

cgyurgyik, 2021-10-20:
> @cgyurgyik any specific use cases to currently motivate this? For example, something like the problems with the exp generator (#505)?

Yep, that's one case. I also believe it will be necessary for testing larger models using fixed point such as `lenet`, or even some of the TVM operations (e.g. convolution, pooling).