calyxir/calyx #792: [Interpreter] NTT-32 failing.

cgyurgyik, 2021-11-13:
```
 fud e evaluations/cidr-pldi-2022/benchmarks/ntt-32.futil --to interpreter-out -s verilog.data evaluations/cidr-pldi-2022/benchmarks/ntt-32.futil.data -pr 
```

Results in:

```
✖ interpreter → interpreter-out
`./target/release/interp  "--no-verify" -l /Users/cgyurgyik/Projects/futil-clone/futil --data /var/folders/92/gt5cgc1d6rz4r1yvkl_p9www0000gn/T/tmp3wakiddi/data.json /var/folders/92/gt5cgc1d6rz4r1yvkl_p9www0000gn/T/tmpxg1y3ds4' failed:
=====STDERR=====
thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', interp/src/interpreter/steppers/group_interpreter.rs:156:61
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

=====STDOUT=====
```

This works on `verilog`, `iverilog` with expected results:

```
{
  "cycles": 272,
  "memories": {
    "a": [
      14,
      50,
      49,
      58,
      48,
      6,
      75,
      69,
      20,
      94,
      65,
      58,
      62,
      9,
      18,
      46,
      29,
      5,
      40,
      40,
      30,
      87,
      90,
      7,
      10,
      12,
      70,
      32,
      74,
      70,
      30,
      14
    ],
    "phis": [
      1,
      28,
      8,
      30,
      64,
      46,
      27,
      77,
      22,
      34,
      79,
      78,
      50,
      42,
      12,
      45,
      96,
      69,
      89,
      67,
      33,
      51,
      70,
      20,
      75,
      63,
      18,
      19,
      47,
      55,
      85,
      52
    ]
  }
}
```

cgyurgyik, 2021-11-13:
https://github.com/cucapra/calyx/blob/161e09471ea9e1198d2f479bce7ff3a7036b5b4a/interp/src/interpreter/steppers/group_interpreter.rs#L155-L157

EclecticGriffin, 2021-11-13:
Investigating now

cgyurgyik, 2021-11-13:
Ok thank you. The program is quite large, and I haven't had a chance to try and minimize it yet.

EclecticGriffin, 2021-11-13:
I've found the bug (it's very silly), trying to resolve a discrepancy in the outputs

PR #793 (2021-11-14): [Interpreter] Patch sdiv mixup
closes #792 

Note there is a discrepancy in the output which may or may not be a bug with something
closed 2021-11-14 by EclecticGriffin