calyxir/calyx #807: The Inlining Pass

rachitnigam, 2021-11-28:
Design document for implementing the component inlining pass that fully inlines the structure and control of a sub-component.

For the following program, we want to inline the two instances of the `increment` component into `main`.

```
component increment(in: 32) -> (out: 32) {
  cells {
    r0 = std_reg(32);
    add = std_add(32);
  }
  wires {
    group init {
      r0.in = in;
      r0.write_en = 1'd1;
      init[done] = r0.done;
    }
    group incr {
      add.left = 32'd1;
      add.right = r0.out;
      r0.in = add.out;
      r0.write_en = 1'd1;
      incr[done] = r0.done;
    }

    out = r0.out;
  }
  control {
    seq { init; incr; }
  }
}

component main(in: 32) -> (out: 32) {
  cells {
    my_out = std_reg(32);
    @inline t1 = increment();
    @inline t2 = increment();
  }
  wires {
    out = t2.out;
  }
  control {
    seq {
      invoke t1(in = in)();
      invoke t2(in = t1.out)();
    }
  }
}
```

The process needs to:
1. Inline all of the cells in the instance.
2. Inline all the wires and groups in the instance.
3. Replace all `invoke` for the component with the control program.

Each step needs to carefully ensure that the structure of the parent component does not get screwed up.

## Inlining Cells

A component can have multiple possible instances that need to be inlined. Each instances needs to separately have all its associated cells be inlined which requires unique names to be generated while the cells are being defined.

We can use the `ir::Component::gen_name()` method to create a new name for each cell. The method ensures that the process of inlining does not generate conflicting names. Additionally, we can add the name of the original instance as a prefix to all the cells being inlined so we have some debuggability.

Because we rename all cells for each instance, we need to make sure that when the wires and control are inlined, they refer to the newly generated names instead of the original cells names in the component definition.

After the cell inlining process, the `main` component look like the following. Note that it is currently malformed because we're visualizing what happens in the middle of a pass:
```
component main(in: 32) -> (out: 32) {
  cells {
    my_out = std_reg(32);
    t1_r0 = std_reg(32);
    t1_add = std_add(32);
    t2_r0 = std_reg(32);
    t2_add = std_add(32);
  }
  wires {
    out = t2.out;
  }
  control {
    seq {
      invoke t1(in = in)();
      invoke t2(in = t1.out)();
    }
  }
}
```

## Inlining Wires

Each instance may have groups and continuous assignments that need to be inlined. We need to take care of two things:

### Groups

Inlining groups is relatively straightforward–we rename the group so that it is unique in the parent component by giving it the instance prefix and using the `ir::Component::gen_name()` method. Next, we rewrite all the assignments in the groups so that they refer to the cells generated in the previous step instead of cells in the component definition of the instance. The internal pointers within the assignment should correctly point to the cells generated in the previous step.

### Internal Continuous Assignments

For continuous assignments to *internal ports*, i.e., ports on the sub-component of the instance, we can just rewrite the assignment to point to the newly generated cells.

### Continuous Assignments to IO Ports

**Needs Discussion**

From the above example, we have:
```
component increment(..) -> (..) {
  ...
  out = r0.out;
  ...
}
component main(..) -> (..) {
  ...
  wires {
    out = t2.out;
  }
  control {
    ...
    invoke t2(in = t1.out)(..)
  }
}
```

The uses of `t1.out` and `t2.out` need to be somehow replaced with instance-specific `r0.out` ports. However, in the general case, there can be many drivers to the `increment.out` port that need to be inlined requiring the duplication of all *readers* of that port. This might be a place where we need to introduce a `let`-binding construct in Calyx that gets compiled down to named wires.

## Inlining Control

Two things need to be handled:
1. Renaming the groups to refer to the instance-specific ones generated in the previous part.
2. Inlining uses of the input--output ports.

### Renaming Groups

Pretty straightforward–rename the component-specific groups to instance-specific groups.

### Parameters

**Needs Discussion**

This is tricky. If the same instance is invoked multiple times, it is going to have multiple drivers to the same port that need to be active when the corresponding invocation would have been active.

For example, in the following program, `t1` is invoked multiple times with different values for its input port `in`:
```
seq {
  invoke t1(in = 32'd0)();
  invoke t1(in = 32'd1)();
}
```
If `t1` has the following usage of `in`:
```
group init {
  r0.in = in;
  r0.write_en = 1'd1;
  init[done] = r0.done;
}
```
We are unable to inline the two invocations. One possible solution to this is introducing the `with <comb_group>` operator for the enable control operator so that we can generate something like:
```
wires {
  comb group in_assign_1 { in = 32'd0; }
  comb group in_assign_2 { in = 32'd1; }
}
seq {
  t1_init with in_assign_1;
  t1_init with in_assign_2;
}
```

cgyurgyik, 2021-11-28:
> This is tricky. If the same instance is invoked multiple times, it is going to have multiple drivers to the same port that need to be active when the corresponding invocation would have been active.

I'm having trouble following this. Why can't we just inline the port values into their respective groups and give them unique names, e.g.

```
group t1_init__0 {
  r0.in = 32'd0;
  r0.write_en = 1'd1;
  init[done] = r0.done;
}
group t1_init__1 {
  r0.in = 32'd1;
  r0.write_en = 1'd1;
  init[done] = r0.done;
}

...

seq {
  t1_init__0;
  t1_init__1;
}
```

rachitnigam, 2021-11-29:
How would you inline this:
```
comb group do_add {
  add.left = 32'd10; add.right = r0.out;
}
control {
  invoke t1(in = add.out)(...) with do_add;
}
```

In general, the port being referred to can be using an arbitrary circuit dataflow before it. Additionally, inlining this way feels inelegant–you have to partially evaluate the control program every time you want to inline an `invoke`.

cgyurgyik, 2021-11-29:
The way I foresee inlining an invoke with a combinational group is adding the wires:
```
add.left = 32'd10; add.right = r0.out;
```
to the respective group(s) that use the value `in`, and let another pass (e.g. constant-fold, code motion) optimize the duplicated stuff. Perhaps less elegant though.

rachitnigam, 2021-11-29:
One implicit constraint to think about is if inlining generates the same hardware or not. For example, the following invokes:
```
seq {
  invoke t1(in = a.out)(...)
  invoke t1(in = b.out)(...)
}
```
Get lowered to:
```
group tdcc {
  t1.in = fsm.out == 0 ? a.out;
  t1.in = fsm.out == 1 ? b.out;
}
```
Which puts a mux in front of the input port of the `in` port of `t1`. However, if you inline every use of `in` within the `t1` control program, it will create a mux *for every use* of `in`:
```
component t1 {
  group one { x.in = in }
  group two { y.in = in }
  control { seq { one; two; } }
}
component main {
  seq {
    invoke t1(in = a.out)(...)
    invoke t1(in = b.out)(...)
  }
}
```
Turns into:
```
component main {
  group one__1 { x.in = a.out }
  group two__1 { y.in = a.out }
  group one__2 { x.in = b.out }
  group two__2 { y.in = b.out }
  control {  seq { one__1; two__1; one__2; two__2; } }
}
```
Which turns into:
```
group tdcc {
  x.in = fsm.out == 0 ? a.out;
  x.in = fsm.out == 2 ? b.out;
  y.in = fsm.out == 1 ? a.out;
  y.in = fsm.out == 3 ? b.out;
}
```
This program has two muxes, compared to one mux in the original program. Also, the generated FSM is bigger (this is unavoidable).

It seems the bigger FSM might be an "expected cost" of inlining while the additional muxing might be surprising and scale non-linearly with the amount of inlining that occurs.

cgyurgyik, 2021-11-30:
Just so I'm understanding correctly: 

```
component t1 {
  group one { x.in = in }
  group two { y.in = in }
  control { seq { one; two; } }
}
component main {
  seq {
    invoke t1(in = a.out)(...)
    invoke t1(in = b.out)(...)
  }
}
```
->

```
component main() {
  wires {
    group one { x.in = t1_in }
    group two { y.in = t1_in }
    comb group in_assign_1 { t1_in = a.out; }
    comb group in_assign_2 { t1_in = b.out; }
  }
  seq {
    one with in_assign_1;
    two with in_assign_1;
    one with in_assign_2;
    two with in_assign_2;
  }
}
```

This only produces a single mux since we have two groups: `one` and `two`. 

I wonder if this will make future software-like passes more difficult to reason about. I haven't thought enough about this, and it may be unwarranted given it is not necessarily the case that basic block <=> group.

rachitnigam, 2021-11-30:
Yup. This requires imbuing Calyx with the ability to define `t1_in` as a wire that can be assigned to and used. I thought the CIRCT dialect already has something similar. Hopefully it doesn't make things harder to reason about because you can "just" follow the use-def chain but multiple assignments to the same port would make things harder.

EclecticGriffin, 2021-11-30:
Right, so if I'm understanding this correctly, the central difficulty is that inlining a sub-component effectively erases the I/O ports of that sub-component because they're not attached to anything which is present in the inlined program. And this has the knock-on effect of making a bunch of things more complicated because we've lost a level of abstraction around how data is supplied to and taken from a sub-component, i.e., we now need to think about everywhere within the sub-component that the data can be emitted from and everywhere outside the sub-component that the data can be supplied from.

This does make it seem like named wires could be useful! Another---perhaps silly---way of thinking about it is the creation of phantom ports which are not attached to anything and can be both written and read, kinda like the existing holes in groups.

Maybe it makes sense to try to preserve the I/O port style abstraction for inlined sub-components, though I don't know if this runs contrary to the goal of inlining as interacting with the sub-component still requires arbitration by this abstraction.

Something else to think about is the explicit use of the go ports of sub-components and whether or not there will be difficulties arising from non-invoke uses. I can't think of any off the top of my head, though, so this might be moot point

cgyurgyik, 2021-11-30:
So when we say named wire, I was imagining:

```
// Represents a physical wire in hardware.
// Constraints: single input, can be spliced to arbitrary number of outputs.
t1_in = std_wire(32);

...

// This is syntactic sugar for t1_in = mux(cond, a.out, b.out);
t1_in = a.out;
t1_in = b.out;
```

In software language, it is an uninitialized variable. Use-before-initialization is error or UB.

rachitnigam, 2021-12-01:
> Something else to think about is the explicit use of the go ports of sub-components and whether or not there will be difficulties arising from non-invoke uses.

I actually don't think we can inline structural uses of components because they can be used in arbitrary groups and we don't really know where to place the inlined control program of the component when they are invoked from a non-trivial group.

For now, this whole proposal only focuses on components that are *only* used through invokes.

> In software language, it is an uninitialized variable. Use-before-initialization is error or UB.

Assignments are always active so the example above looks wrong–it has two non-conditional drivers active. Also, I don't follow this analogy. If assignments are always active, there is no use-before-initialization.

cgyurgyik, 2021-12-01:
The `cond` is supposed to be the FSM state above with the two assignments in separate groups, I just didn't clearly spell that out. 

The use-before-initialization is more so saying if `t1_in` is used before it is defined, then it is probably not what the user wanted. By used before defined, I mean it shows up on the RHS of an assignment before it shows up on the LHS while walking the control.

rachitnigam, 2021-12-01:
I wasn't imagining you could observe the value of a wire outside of a group if that's what you mean. If not, then the port is "just" undriven in the same way any other combinational primitive can be. For example, if you don't drive the `left` and `right` ports of a adder at the same time, the `out` port is undriven and reading it is undefined.

rachitnigam, 2021-12-29:
Reopening this to discussion future implementation questions.

PR #829 (2021-12-28): Inlining pass
Fixes #807. This implements a restricted version of the pass discusses in the issue and requires that instances that are going to be inlined are always invoked with the same parameter bindings. Surprisingly, this is already pretty good---we can fully inling TCAM and systolic arrays. In the future, we can extend this pass to support `invoke` of the same instance with different parameters but that requires implementation of the `with` operator that can activate a combinational group in a given scope.

## Implementation
- [x] Handle parameters
- [x] Handle reading from output ports
 [... cut]
closed 2021-12-29 by rachitnigam