calyxir/calyx #812: [Tracking] Frontend data format audit

EclecticGriffin, 2021-12-02:
This is a tracking issue to audit the types of data being used by the frontends to inform the future development of custom data formatting in the interpreter and debugger.

Frontends:

- [ ] Dahlia
- [ ] Systolic
- [ ] Relay
- [ ] NTT
- [ ] MrXL

rachitnigam, 2021-12-15:
Just to clarify, this is not about data format but data structures right? I remember this came out of a conversation about supporting source-level data structure debugging.

EclecticGriffin, 2021-12-16:
That's right, should probably rename it to data structure audit

rachitnigam, 2022-04-18:
@EclecticGriffin is this still an action item for the interpreter/paper submission?

EclecticGriffin, 2022-04-18:
This is not immediately relevant to the current submission target. We could consider this something of a stretch goal if we decide there is sufficient time to handle data structure variable translation and data structure reconstruction, but that seems unlikely for the current submission

rachitnigam, 2022-10-01:
@EclecticGriffin are we still planning to work on this? If not, let's close

rachitnigam, 2022-11-13:
I assume we're not going to work on this so closing the issue

closed 2022-11-13 by rachitnigam