calyxir/calyx #813: [Interpreter] De-libaryify the interpreter

EclecticGriffin, 2021-12-02:
The interpreter/debugger is currently organized as a very time main program which calls into the library implementing all the actual features. Unlike the compiler infrastructure it's not clear that the interpreter should really be exposing a library at all. Probably the crate should just be a bin and not a lib + bin together, though I'm happy to hear contrary opinions.

sampsyo, 2021-12-02:
Here is one very silly reason to consider keeping the library! It would make it easier to make alternative frontends, like an interactive web GUI that works by compiling the library to WebAssembly (like we do with the compiler). Not sure we’ll ever do that, but it would be neat!

EclecticGriffin, 2021-12-02:
Ah fair, that does make sense!

EclecticGriffin, 2021-12-03:
Going to close this since I no longer think it's necessary

closed 2021-12-03 by EclecticGriffin