calyxir/calyx #815: [Interpreter] Verbose trace info / more robust logging

EclecticGriffin, 2021-12-03:
It might be useful to add more logging capabilities to the interpreter since we've added slog. This could make future internal debugging a bit easier and could potentially offer more insight for program execution, though by default these logging messages will not be present in the release mode (which means they're essentially free in some respects).

I think adding these and a verbose / very verbose flag to the interpreter could be helpful, though far from critical

cgyurgyik, 2021-12-04:
An idea I was pondering... the infamous print statement debugging. Add LOG statements directly into the Calyx program, e.g.,
```
group G {
  add.left = a.out;
  add.right = b.out;
  result.in = add.out;
  /// LOG_INTERPRETER("$0 + $1 = $2", add.left, add.right, result.in)
}
```

Each time group G is executed, this is logged.

rachitnigam, 2021-12-15:
> An idea I was pondering... the infamous print statement debugging. Add LOG statements directly into the Calyx program, e.g.,

I don't hate this idea but any reason why this should be in the source program and not just a command in the debugger?

cgyurgyik, 2021-12-15:
Simplicity, ease of use, print statement debugging is nice when you want to quickly iterate between changes and their effects.

rachitnigam, 2021-12-15:
Let's talk about this specific example: how often should I expect the log statement to execute? Every time the value of one of the ports mentioned in it (which makes it equivalent to `$monitor` in Verilog) or every time the group finishes one clock tick, or once the group is `done`?

cgyurgyik, 2021-12-15:
Yeah I was thinking about that as well, and I'm not sure what the best answer is. I think every clock tick within the group is a good starting point, with perhaps an option to minimize this just when `done` is true.

rachitnigam, 2022-03-10:
#815 adds the `tracing` crate to the IR library so this should be possible now?

EclecticGriffin, 2023-05-01:
Closing this for the moment. Will re-open in the future if logging features become more relevant

closed 2023-05-01 by EclecticGriffin