calyxir/calyx #818: [Repo] Add Cargo.lock to the repo

EclecticGriffin, 2021-12-13:
Following the minor issues that came up when merging #816 it seems like we should probably commit the lock file to avoid this in the future (following a conversation with @sampsyo). 

Tagging @rachitnigam since this particular decision predates my involvement so I don't know the underlying reasons.

sampsyo, 2021-12-14:
Seems like the thing to do! As I understand it, the recent-ish trend toward committing "lockfiles" for package manager is essentially to avoid CI surprises like the one referenced above.

rachitnigam, 2021-12-15:
Seems like Rust people recommend it for binaries: https://doc.rust-lang.org/cargo/faq.html#why-do-binaries-have-cargolock-in-version-control-but-not-libraries

commit 3338a72a (2021-12-17): add Cargo.lock. Fixes #818
closed 2021-12-17 by rachitnigam