calyxir/calyx #840: Optimization: Convert static `par` of `seq` to `seq` of `par`

rachitnigam, 2021-12-28:
This transformation is in the same spirit as #634. For the following program:
```
par {
  seq { A0; B0; C0; }
  seq { A1; B1; C1; }
  seq { A2; B2; C2; }
}
```
We'll end up generating three FSMs, each with three states. On the other hand, this program:
```
seq {
  par { A0; A1; A2; }
  par { B0; B1; B2; }
  par { C0; C1; C2; }
}
```
we generate one FSM with three states. However, in general, we cannot transform program one into program two because we do not know the timing behavior of the program.

Given a program on with inferred or specified timing behavior, we can perform the transformation:
```
par {
  seq { @static(M) A0; @static(N) B0; @static(P) C0; }
  seq { @static(M) A1; @static(N) B1; @static(P) C1; }
  seq { @static(M) A2; @static(N) B2; @static(P) C2; }
}
```

The one thing to note however, is that the last program, if compiled with the (completely broken) static timing pass, the generated FSM will essentially have three states. (This is also true for #634). Maybe these optimizations are just a way to approximate the FSMs generated by `static-timing` anyways.

However, I'm sure that these transformations can occur recursively into other control operators and provide similar benefits. And I think with control programs where larger fragments are statically timed, we can perform more complex program reshaping to enable optimized designs.

sampsyo, 2022-01-03:
A cool thought! Sorry to be a broken record about this, but I think transformations like this must really depend on a better development of the semantics of `par` and timing more broadly. WIth that in place, it would be possible to build a more crisp argument about whether and when such a transformation is legal: i.e., the transformed program’s semantics refine those of the original program. For example, it seems likely to be true that this is safe under our current "communication not allowed" semantics but more difficult to justify under any communication-allowed semantics.

calebmkim, 2022-06-07:
First Thoughts/Steps: 
1) Add a new pass that does nothing except prints "hi" and make sure it works
2) (Now start actually working on the pass) Make sure the original par has only seqs as its statements 
3) Create a new seq. 
4) Construct a new par containing the 1st statement in each of the original seqs (i.e., the seqs in the original par). If we find that these statements do not all take the same number of cycles, then stop. 
5) Add this new par to the end of the new seq we just constructed. Repeat this process. At any point if we find out that the original seqs are actually of different lengths, then stop. 
6) Replace the original par with the new seq we just constructed. 

Other thoughts 

Once we get the most conservative way going we can generalize the pass more. For example, we may not want to force all the seqs to be the same length. If we have 
```
par {
  seq { @static(M) A0; @static(N) B0}
  seq { @static(M) A1; @static(N) B1; @static(P) C1; }
  seq { @static(M) A2; @static(N) B2; @static(P) C2; }
}
```
Maybe this could become: 
```
seq {
  par { @static(M) A0; @static(M) A1; @static(M) A2;}
  par { @static(N) B0; @static(N) B1;  @static(N) B2; }
  par { @static(P) C1; @static(P) C2 }
}
```

There are also possibilities for when the statements in each seq do not take the same number of cycles. 
If we have 
```
par {
  seq { @static(M) A0; @static(N) B0; @static(P) C0}
  seq { @static(Q) A1; @static(R) B1; @static(S) C1; }
}
```
Maybe this could become: 
```
seq {
  par { @static(max(M,Q)) A0; @static(max(M,Q)) A1 }
  par { @static(max(N,R)) B0; @static(max(N,R)) B1 }
  par { @static(max(P,S)) C0; @static(max(P,S)) C1 }
}
```
However, looking at this, it may not be the best idea. Since it will now take max(M,Q) + max(N,R) + max(P,S) cycles instead of max(M + N + P, Q + R + S).

rachitnigam, 2022-06-18:
Fixed by https://github.com/cucapra/calyx/pull/1026

closed 2022-06-18 by rachitnigam