calyxir/calyx #859: Parallel assignment asserts do not work with icarus verilog

rachitnigam, 2022-01-07:
https://github.com/cucapra/calyx/pull/847 attempted to enable parallel assignment checking with the icarus backend but inadvertently made all tests fail (another argument for #755). The problem is that the `onehot` primitive returns false for constants with `000x` where `x` is generated from unassigned ports causing the program to fail.

Additionally, because the assertions are always active, even during cycles responsible for resetting undriven ports, the program will immediately fail. The likely fix is to add an additional guard in front that ensure that checking only occurs when reset if false. 

Sadly, this is not as simple as `~reset & ~onehot(...)` because `0 & x = x` in Verilog which is really really unfortunate.

sampsyo, 2022-01-07:
Very very tricky. I wonder if there's any way to configure Icarus to just throw errors the first time "x" appears instead of propagating "x"… seems unlikely, but just thought it might be worth poking around to see.

rachitnigam, 2023-04-29:
One possible fix to this is using `===` which allows checking whether a value is `'x` or not. We can generate the following conditional:
```
if (g1 !== 'x & g2 !== 'x ... ) begin
  if (~onehot(...)) { ... }
```
I'm hoping that the effect of this is that we only execute the onehot check when the guards have defined values.