calyxir/calyx #878: [Tracking] Interpreter Optimization Ideas

EclecticGriffin, 2022-01-24:
## Parallelism

1. Execute par blocks in parallel
2. Parallel execution of assignments (in group execution)

The first of these is more likely to be useful while the latter might not always be faster.

## Performance tuning
3. Avoid executing cells which have not had a value change (during combinational convergence loop)
4. Use dataflow ordering to evaluate the group directly rather than iterating until combinational convergence. (This requires replacing the convergence loop with something else).

There are definitely other things and I will update this issue as I remember them and link the appropriate issues for particular implementation proposals.

rachitnigam, 2022-11-13:
The tracker is about a year old so let's discuss and update it or close it

EclecticGriffin, 2023-05-01:
Closing: stale and not up to date with current efforts

closed 2023-05-01 by EclecticGriffin