calyxir/calyx #882: [CIDR] Improve `where` printing.

EclecticGriffin, 2022-01-26:
The current printing is somewhat hard to make sense of, some improvements would be:
- [ ] Add some colour to the output to distinguish the types of things going one
- [ ] Add line info to invoke statements in the tree
- [ ] Possibly separate out the control leaves and the subcomponents

sampsyo, 2022-01-26:
Just copying a comment I made on Slack, for better organization:

> I can imagine two options:
> - Keep the mixed tree, and add some kind of symbol to denote the two types???
> - Split into multiple views somehow. Like, one view for the static component tree; another for the dynamic tree of active components (which is a subset of the former). Either one could augment each node with its active control leaves, but those would be part of the node (rather than being nodes in themselves).

rachitnigam, 2022-04-12:
@EclecticGriffin has there been progress on this and is this important for the paper?

EclecticGriffin, 2022-04-12:
One could argue the rudimentary source position contributes but this is more about the native calyx program counter business. But in terms of the actual display tree, no new progress. I'm not sure that this is really useful for the paper, though I'm happy to be persuaded otherwise.

EclecticGriffin, 2023-05-01:
Closing this for the moment as I am not prioritizing debugger front-end stuff right now. Would like to do something for this eventually though

closed 2023-05-01 by EclecticGriffin