calyxir/calyx #889: [Tracking] Calyx Fuzzing System

crystalhu26, 2022-02-01:
**Goals**

1. Prove equivalence or in-equivalence of calyx programs by comparing outputs with same inputs
2. Find bugs in calyx compiler (input fuzzing)
3. Find bugs in the backends (output fuzzing)

**Overview**
The fuzzing system will show equivalence or in-equivalence of 2 input files by comparing their outputs with the same input data. There are two major parts of the fuzzing system: input files and output comparison. 
[image]

1. Input fuzzing: inputs include two files and a data template. 

- Two input files (`input_file_1`, `input_file_2`) can be either calyx file or some generic files such as dahlia and systolic that are used to determine equivalence. These two files will be transformed into calyx files (`calyx_file_1`, `calyx_file_2`) and feed into fuzzing system.
- To restrict the testing data format, a data template (`data_template`) will also be provided to indicate data type and shape. After input files being transformed into calyx, data will be generated based on data template and feed into the fuzzing system. The data template can be further improved by implementing an auto data generator that infer data type and shape from input files. 

2. Output fuzzing: outputs comparison. 

- The outputs will be transformed into same states for comparison using tools such as Verilator, icarus, and cidr. (tool)
- Two outputs (`output_file_1`, `output_file_2`), in JSON format, correspond to two input files and will be employed for comparison.

**Milestones**

1. Build tool that requires `calyx_file_1`, `calyx_file_2`, and a backend tool to compare.
2. Accept inputs in any format (such as transforming dahlia to calyx) and feed into fuzz.
3. Accept backend tools in any format to transform fuzz to fuzz output.
4. Implement the data_gen for data template.

rachitnigam, 2022-02-01:
Thanks for writing this up @crystalhu26! The first step to get started would be writing up an [external stage](https://docs.calyxir.org/fud/external.html) in `fud` to implement the fuzzing step. As discussed, we can start off by hard coding the requirement that the specification file needs to be in Calyx and eventually use `fud`'s magic to compile specification files in different states.

rachitnigam, 2022-10-01:
I don't see us working on the fuzzing system anytime soon so closing this for now!

closed 2022-10-01 by rachitnigam