calyxir/calyx #891: [Interpreter] If, While, and the relationship between `step` and clock ticks.

EclecticGriffin, 2022-02-02:
Making this to supersede #883.

The high level question here is "What do we want the relationship between an interpreter `step` and the clock-tick of a hypothetical design to be" which I'm going to poke at via discussing If and While.

The way I see it there are two ways the interpreter can treat conditions for if/while statements which I am going to call:
1. eager
2. lazy

This has to do with whether we evaluate the conditional the moment we reach an if/while (on sub-interpreter creation) or if we wait until the interpreter is explicitly advanced to evaluate the condition.

Currently the interpreter operates eagerly for if/while statements without a comb group. The moment we reach one of these statements it evaluates the conditional meaning that the execution path has already been determined before the first `step` call to this sub-interpreter. 

For if/while statements with a comb group, the condition is evaluated lazily. The condition is not touched until after the first call to `step` (it actually takes multiple calls to step to fully branch based on the condition hence #883).

The reason for this set up is because it was previously possible for the group attached to if/while statements to be non-combinational so the interpreter would not make any assumptions about the timing behavior of the group. However as this is no longer the case we now need to consider what the "correct" behavior should be which is really a matter of what we expect the relationship between `step` and the clock to be.

---

I think a state outline will help elucidate the differences.

0. interpreter exists but has done nothing so far.
1. evaluate the condition
2. evaluate the appropriate body 
3. done.

Eager evaluation means that before calling `step` the interpreter will have already moved from stage 0 to stage 1 to stage 2 (or 3 if a while group with false) and the `step` invocation will advance the interpretation of the body (if applicable).

Lazy evaluation means that before calling `step` the interpreter is in stage 0 and will be moved into stage 1 (currently requiring further invocations to move to stage 2).

---

I believe there is a potential bug/semantic grey area with respect to eager evaluation and conditionals which directly depend on some input to the component. So supposing a component executing an if statement whose conditional depends on the input (other than go) we can potentially get into a strange situation if the input of interest varies exactly at this transition (or if the component is paused and resumed with different input) as an eager evaluation would effectively miss an external computational update for conditions w/o a `with` clause. (apologies if this doesn't make sense, I am having trouble finding the right words).

There is also a philosophical question as to how many `steps` should be required to evaluate the condition. 

let (=>) represent a single invocation of `step` and let (->) represent intra-step state transitions. Here are the potential options for thinking about how this should work.

A.
```
0 => 1 => ... => 2 => ... => 3 (Lazy, stepped cond eval)
```

B.
```
(0 -> 1 -> 2) => 2 => ... => 3 (Eager eval, instant cond eval)
```

C.
```
0 => 1 -> 2 => ... => 3 (lazy, instant cond eval)
```

D (weird hypothetical)
```
(0 -> 1) => 1 => ... => 2 => ... => 3 (eager, stepped cond eval)
```

A is how the interpreter currently behaves for while/if statements with an attached combinational group while B is how it behaves for those without. This might be undesirable as it makes it harder to reason about time passing.

## Eager/Lazy

Why might we prefer eager evaluation? Eager evaluation has the benefit of requiring less state information to be stored and is possibly more "honest" with respect to the timing of the accelerator that Calyx generates. Though it does suffer from the potential bug listed above.

## Intra-step cond eval vs inter-step cond eval

Instant evaluation of conditionals requires fewer invocations of `step` to advance the program and is more "honest" for cases without an attached combinational group, though it is potentially less "honest" (timing wise) for groups with an attached combinational group as it will really take a full cycle to ensure that the result of the group is stored so that the condition is stable.

The opposite holds for inter-step conditonal evaluation.

---

We can also ask a similar question for `seq` blocks. When we perform the last `step` for an element of the sequence are we immediately moved to the state where we are executing the next element or is there an interstitial inactive state to be stepped through?

... => 1 => 2  => ... 

vs

... => 1 => X => 2 => ...

Anyway apologies for this long mess, hopefully some of the distinctions I am trying to draw here make a bit of sense. Please let me know if you have thoughts or questions.

rachitnigam, 2022-02-04:
> Currently the interpreter operates eagerly for if/while statements without a comb group.

Just to clarify, when we have a program:
```
one;
if lt.out {
  two;
}
```

The interpreter will evaluate `lt.out` in the environment generated by `eval(one)` or in `reset(eval(one))`? 

> it actually takes multiple calls to step to fully branch based on the condition

I don't follow this. Combinational groups should take only one combinational step before needing to being evaluated right? Is `step` lower level than my notion of a "combinational step"?

-----

It might be helpful to explain the step examples above using a concrete Calyx program.

EclecticGriffin, 2022-02-04:
To answer your questions:

> The interpreter will evaluate lt.out in the environment generated by eval(one) or in reset(eval(one))?

The interpreter will look at `lt.out` under `reset(eval(one))` as it is not doing any sort of scoped reasoning and the language semantics no longer make using `lt.out` reasonable without a `with` clause as it's not stable.

> I don't follow this. Combinational groups should take only one combinational step before needing to being evaluated right? Is step lower level than my notion of a "combinational step"?

Your intuition is correct, this has to do with the current imp still believing non-combinational groups are possible in a `with` clause. Regardless of the overall direction we want to take, this piece should still be fixed.

PR #933 (2022-02-25): [Interpreter] Revamp the stepper state machines and other minor changes
Closes #891. Mostly this just cleans up the state machine transitions in the interpreter so that the guarantees are explicit rather than the implicit conjunction of optional states. For Rust reasons, this also means I had to add a bunch of `unreachable!()`s which is depressing but hopefully isn't too challenging to parse.

Also I discovered `where Self:Sized` is a trait bound you can write which is neat!

We can revisit #891 if we want to change the precise step dynamics, but otherwise I consider it resolved.
closed 2022-02-25 by EclecticGriffin