calyxir/calyx #896: `comb-prop`: Inlining reads from wires

rachitnigam, 2022-02-03:
When faced with an invoke of this form:
```
component p()(x: 32) {
  group do_x {
    x = foo.out;
  }
}
invoke p()(x = x.in);
```
The `inline` pass will generate assignments like this in the parent component:
```
group do_x {
  p_x.in = foo.out;
}
x.in = p_x.out;
```

We should be able to inline the *read* from the wire `p_x.out` with a write into `x.in`:
```
group do_x {
  x.in = foo.out;
}
x.in = p_x.out;
```

Stating somewhat formally–unconditional writes into ports from wires can be inlined into the context where the write to the wire occurs. I think this works even if `p_x.in` is written to from multiple groups but need a sanity check from @sampsyo and @EclecticGriffin on this.

sampsyo, 2022-02-04:
Yo yo, I'm a tad bit confused about the first example here (probably just low-level stuff, but bear with me):

* It looks like `x` is being used as an "out" port to `p`, right? Because it's assigned to in the group. (But the signature has it in the "in" position.)
* But… it seems like the `invoke` uses `x` as an _in_ port for `p`, so maybe the above is wrong??
* Is the `x` that's the parameter to `p` the same as the `x` used in `x.in`? Probably not, right—that's just some other cell with the same name as the port?

Taken together, this makes me somewhat confused about what the transformed (last) listing is supposed to do… it naively looks like `p_x.out` is read without ever being written? Sorry to be so confused, but I don't quite get the insight yet…

rachitnigam, 2022-02-04:
Ah, sorry for the sloppy example, I've tried to fix it. Let me know if it makes more sense. At a high-level, here is the use case:
- There is a subcomponent `p` with an *output* port called `out`.
- The parent component `invoke`s `p` and connects the input of another component, `y` to the output from `p`. This intuitively means that whatever output is generated on `out` during the execution of `p`, it should flow into `y.in`.

Now, when we inline the component `p`, we need some way to represent this output-input connection. First, we generate the wire `p_out` to represent `p.out`. Next, we define the connection `y.in = p_out.out` to state that whatever output is generated from `p.out` should flow into `y.in` (of course, this is not quite correct because the assignment is scoped and shouldn't always be active. This is a bug and I'll open a different issue for this).

Now, onto the actual pass. If we have a connection like this (forgetting for a moment that inlining is doing the wrong thing by not scoping it):
```
y.in = p_out.out; // p_out is a wire
```
It means that any use of `p.in` in the program can be replaced with `y.in` because the continuous assignment would've immediately forwarded to `y.in` anyways.

Hence, the optimization: if there is an unconditional continuous assignment from the *output port* of a wire to a subcomponent's *input port*, replace all writes to the *input port* of the wire with the input port of the subcomponent. 

Does that make more sense?

rachitnigam, 2022-02-04:
When we have multiple reads to outputs:
```
group do_x {
  p_out.in = foo.out;
}
y.in = p_out.out;
x.in = p_out.out;
```
turns into:
```
group do_x {
  y.in = foo.out;
  x.in = foo.out;
}
```

EclecticGriffin, 2022-02-04:
So if I'm understanding correctly this replaces the continuous assignment with a direct one in the group? This won't always be possible right, only when there's a single consumer for a given sub-component?

rachitnigam, 2022-02-04:
> This won't always be possible right

See the example in the comment above–you need to inline all consumers of the output port.

EclecticGriffin, 2022-02-04:
Ah fair, and the reason we don't do this by default is because it's hard to manage during the inlining process?

rachitnigam, 2022-02-04:
No, it's because its not the job of the inlining pass to cleanup the generated pass. Or rather, this logic is separate from inlining.

EclecticGriffin, 2022-02-04:
fair enough

sampsyo, 2022-02-04:
Thanks for clarifying a bit! This makes sense to me. The "multiple reads" things is a tiny bit tricky, but I think the idea to treat "continuous assignments to wire components" as a propagation opportunity seems pretty clear. Nice!

PR #897 (2022-02-09): Propagate reads from wires
Fixes #896. It inlines continuous reads from wires when there is exactly one such use of a wire. If there are multiple reads from the same wire, it bails out for that assignment.

A future PR may handle the multiple read case.
closed 2022-02-09 by rachitnigam