calyxir/calyx #910: Proposal: Make `wire-inliner` default strategy for compiling holes

rachitnigam, 2022-02-11:
The current compilation pipline uses the `hole-inliner` pass to compile away holes generates during the compilation process. We've added the `wire-inlining` pass that instead generates wire assignments for all hole assignments which has the benefit of making the final program generated be smaller. It also ensures that programs won't run into the problem of having lots of really long assignments which can be a problem for Verilator.

The original rationale for `hole-inliner` was maybe it exposes more opportunities to simplify the guard expressions but it seems that this kind of optimization is best left to lower-level compilers in the stack.

sampsyo, 2022-02-14:
Super interesting! Maybe this is going too far, but is it a corollary that holes should be semantically identical to `std_wire` instances?

rachitnigam, 2022-02-14:
Yup, holes are the same as wires. Maybe we should think harder about the implication from this. It seems that groups basically expose two "control handles": `done` to terminate their own execution and `go` to signal when they start.

PR #937 (2022-02-25): switch to wire-inliner for lowering
Fixes #910.

The `top-down-st` pass is generating massive assignments so `hole-inliner` chokes up. This makes `wire-inliner` the default strategy for lowering which is much faster and simpler.
closed 2022-02-25 by rachitnigam