calyxir/calyx #919: Intepreter incorrectly computes values when there is a par dependency

rachitnigam, 2022-02-17:
I was expecting the lockstep semantics to do the right thing but I guess it doesn't. Here is the program:
```
// -p validate -p par-to-seq
import "primitives/core.futil";
component main() -> () {
  cells {
    @external r0 = std_mem_d1(32, 1, 1);
    @external r1 = std_mem_d1(32, 1, 1);
  }
  wires {
    group do_r0 {
      r0.addr0 = 1'd0;
      r1.addr0 = 1'd0;
      r0.write_en = 1'd1;
      r0.write_data = r1.read_data;
      do_r0[done] = r0.done;
    }
    group do_r1 {
      r1.addr0 = 1'd0;
      r1.write_en = 1'd1;
      r1.write_data = 32'd30;
      do_r1[done] = r1.done;
    }
  }
  control {
    // After running, we expect:
    // r1 = 30;
    // r0 = r1;
    seq {
      par {
        do_r1;
        do_r0;
      }
    }
  }
}
```

Commands:
```
% fud e --to dat -s verilog.data data.json reorder.futil
{
  "cycles": 4,
  "memories": {
    "r0": [
      20
    ],
    "r1": [
      30
    ]
  }
}
```

and,
```
% fud e --to interpreter-out -s verilog.data data.json reorder.futil -s interpreter.flags ' --allow-par-conflicts'
{
  ...
  "memories": {
    "main": {
      "r0": [
        30
      ],
      "r1": [
        30
      ]
    }
  }
}
```

Note that the `--allow-par-conflicts` is only because of the writes to the `r0.addr0` and an equivalent program with just registers and no `--allow-par-conflicts` will similarly be wrong.

rachitnigam, 2022-02-17:
Actually, this is a bit more tricky to claim as a bug because the semantics of `par` are undefined. @sampsyo (rightly) points out that this kind of assumption, that `do_r0` and `do_r1` will start executing at the same time is a form of "communication" and our current, no-guarantees semantics, do not really allow for this.

EclecticGriffin, 2022-02-17:
Can you paste your data file, I think I know why this is happening but would like to confirm

rachitnigam, 2022-02-17:
AFK but r0 gets 10 and r1 gets 20

EclecticGriffin, 2022-02-17:
Ah yes, I see what's happening. If you switch the order in the par you'll get what you want. This is because lock-step doesn't mean in-parallel. As the interpreter is single-threaded, while each arm is stepped, this does not happen simultaneously. The implementation currently just steps each arm of the par in sequence since there isn't actually any timing guarantee.  In this case, r1 gets to step first and commit its updates followed by r0. It would probably be more accurate to describe the par implementation as a striped lock-step instead since the groups are never running at exactly the same time.

I have some strong reservations about this interpretation of the semantics since it kinda kills modular reasoning about groups. Supporting it would also require fundamentally reworking quite a bit of the interpreter since until this point we've operated on the separate world hypothesis and this makes determining if a port is un-driven a non-local question. I also worry that it commits us to strong timing guarantees for the other operators, since they would be required to make a timing guarantee for par that is meaningful.

EclecticGriffin, 2022-02-17:
Though I suppose the above is a long way of saying that the interpreter's treatment of par up until this point, based on our separate world hypothesis, would argue that the above program is non-deterministic and both outputs are valid. But this is largely because said hypothesis would consider this behavior undefined.

Whereas a single group handling these assignments would produce the "correct" value since there's an implicit timing guarantee amongst the assignments of a group.

EclecticGriffin, 2022-02-17:
I should also note, before I forget, that this points to a current hack in the implementation which is that cell state is shared across branches while port values are not. This is questionable and is probably contributing as well. Excited to talk through some of this fun stuff!

sampsyo, 2022-02-18:
> I have some strong reservations about this interpretation of the semantics since it kinda kills modular reasoning about groups.

Right! That's the trouble I got at, in a much more long-winded way, in #921. I think you've hit the nail on the head here.

> I should also note, before I forget, that this points to a current hack in the implementation which is that cell state is shared across branches while port values are not. This is questionable and is probably contributing as well. Excited to talk through some of this fun stuff!

Also exactly right! It seems like a cool role the interpreter _could_ play is in detecting conflicting writes to registers—in the same way that it currently detects conflicting writes to ports. That seems way more complicated but would have the happy result of catching _all_ bad/nondeterministic/undefined `par` behavior instead of just some of it.

rachitnigam, 2022-02-18:
> As the interpreter is single-threaded, while each arm is stepped, this does not happen simultaneously.

I thought each branch of the par block gets the same environment and the writes are eventually reconcile (which would fix the program above).

> this points to a current hack in the implementation which is that cell state is shared across branches while port values are not.

oh, that explains it then

> I also worry that it commits us to strong timing guarantees for the other operators, since they would be required to make a timing guarantee for par that is meaningful.

Agreed

I'm also kind of surprised that the interpreter doesn't run into this problem with the systolic array generator. Not sure why that is.

EclecticGriffin, 2022-02-18:
Yeah the current hack predates multi-component stuff which unfortunately severely complicates the notion of duplicating environments since we would need to fully duplicate the subtree of a given cell. To say nothing of what merging state looks like in this scheme, All this further suggests a need to rethink the interpreter's current environment set up, but we already knew that

closed 2022-02-18 by rachitnigam