calyxir/calyx #929: Static Timing could generate simpler FSM transition logic.

mikeurbach, 2022-02-22:
In #909, @rachitnigam [pointed out](https://github.com/cucapra/calyx/pull/909#discussion_r811526931):

> It occurs to me that there might be a simple optimization possible in the realize_schedule method to calculate consecutive "runs" of FSM states where we basically just want fsm.in = fsm.out + 1 and simplify the generated transitions a bit. Not super crucial but will probably start mattering when you look at resources.

This would be neat. Instead of a bunch of conditionals and assigning the next FSM state to a specific number, we could simply increment the FSM state counter. We'd need to identify consecutive "runs" of FSM states where this is valid, and generate different hardware. This should be a non-functional change, but we should be able to measure improved resource utilization post-synthesis.

mikeurbach, 2022-02-22:
I didn't know about this test, but I just got notified it is failing because we are generating some truly massive lines of verilog: https://github.com/cucapra/calyx/runs/5282717927?check_suite_focus=true#step:24:472

So besides the above comment, we also [discussed](https://github.com/cucapra/calyx/pull/909#discussion_r811360556) not just counting one giant FSM all the way up:

> The only implementation question is if you want to reset the fsm state to the first state in the body and have a separate counter (to avoid blowing up the main FSM) or just using the original FSM to count up.

This might help cut down on the generated verilog in the failing test...

rachitnigam, 2022-02-22:
Ah, looks like we're also getting some stack overflows: https://github.com/cucapra/calyx/runs/5282717927?check_suite_focus=true#step:24:2590

The tests come from here: https://github.com/cucapra/calyx-evaluation/tree/master/benchmarks
(running `runt -i Correctness` will run the tests). These are bigger Dahlia programs. I assume the stack overflow comes from hole-inliner but top-down-st could also be to blame. I'll try to get a minimal reproduction for the program.

rachitnigam, 2022-02-22:
Okay, it's pretty much the `bound` optimization that is causing all of the problems by generation massive FSM transitions.

Here is an example program that generates tons of transition statements:
[3mm.futil.txt](https://github.com/cucapra/calyx/files/8117024/3mm.futil.txt)

The `hole-inliner` pass overflows with the program but you can dump out the result of `top-down-st` and see the big FSM.

rachitnigam, 2022-02-22:
Just to answer the question from the previous PR, the reason to have a separate counter is that if you use the main FSM for counting, you'll have `l1 * l2 * ... * s` where `l1` is the outermost loop bound, `l2` is the next one, and so on while `s` is the states of the bound. If you have separate counters, you'll have `l1 + l2 + ... + s` states.

mikeurbach, 2022-02-22:
That makes sense. I was initially thinking this isn't the worst, since even for this example with ~16,000 states, we'd only need 15 bits of FSM state, but clearly the verilog we are generating here is not great for tools to work with.

Seems like there are two things we can do:

* Improve the generated low-level calyx to increment a counter through the states rather than having an assignment like `fsm.in = fsm.out == 15'd15590 ? 15'd15591` for each state.
* Split out a separate FSM for each loop. There's probably an interesting resource sharing problem here, but for simplicity I think we'd just want to keep it simple and use 1 per loop

Is there a preference or any other ideas to try? I am thinking it would be good to start with the first approach, see if that improves the generated verilog in practice, and then explore the second approach.

Also, if this is causing issues, please feel free to revert and I will include whatever we come up with when I re-land it.

rachitnigam, 2022-02-22:
I think the first thing to try out is computing runs of states where the transitions happen unconditionally and generating something like:
```
fsm.in = fsm.out > k & fsm.out < p || fsm.out > k & fsm.out < p ... ? incr.out
```

My long-term worry is that even this will run into problems with extremely long guards for the increment logic but it will at least fix our current problem. Once we have this in (and it works), we can look into separating out a counter.

> For the first approach, I know we use some sort of graph library. 

Not sure, my first thought would be to just look at `enables` in the `Schedule` where  the guard [`is_true()`](https://docs.calyxir.org/source/calyx/ir/enum.Guard.html#method.is_true) and computing consecutive runs by sorting a vector of `Range` elements.

closed 2022-02-25 by rachitnigam

rachitnigam, 2022-02-25:
hm, looks like this is not enough to fix the problem still because the generated FSM will trigger the same group many times causing the assignment expression to be massive unfortunately. Looks like we'll need to switch to another encoding for the `@bound` optimized `while` loops to make things work (i.e. have a separate counter).

mikeurbach, 2022-02-25:
That sounds good to me. I will mention if I get a chance to work on it.

mikeurbach, 2022-02-26:
Also- would it be possible to do something similar to #930 but for the assignment guards. I recall they sometimes looked like `(fsm_out >= 1 & fsm_out < 2) | (fsm_out >= 2 & fsm_out < 3)` etc. Maybe we could do a similar collapsing of consecutive ranges there? I haven't taken a look at what the latest looks like for the problematic test cases, just wanted to jot this down before I forget.

rachitnigam, 2022-02-26:
Yeah, there seems to be some optimization work that we can do in making guards simpler. The [MergeAssigns](https://docs.calyxir.org/source/calyx/passes/struct.MergeAssign.html) pass implements some simple heuristics but nothing like this.

PR #930 (2022-02-25): Compress static FSMs
Fixes #929