calyxir/calyx #963: [???] Cholesky Benchmark failing on Verilator 4.028

EclecticGriffin, 2022-03-30:
This is happening on Havarti and its unclear quite what is going on. Since it doesn't seem to reproduce locally it might just be an issue with the old version of verilator running on the machine.

rachitnigam, 2022-04-12:
Can you paste the exact error message?

EclecticGriffin, 2022-04-12:
It's a disagreement between the golden and computed values of the sqrt. I'll go find the exact message

EclecticGriffin, 2022-04-12:
```
[135] %Error: tmpaefd1okc:118: Assertion failed in TOP.main.sqrt0:
sqrt: Computed and golden outputs do not match!
input: 429496782
expected: 0
computed: 65535
```

rachitnigam, 2022-04-12:
Seems like an int max overflow

EclecticGriffin, 2022-04-12:
it does indeed! I assume updating Havarti's verilator will probably fix this I just haven't gotten around to looking up the build from source instructions again

EclecticGriffin, 2022-04-13:
closing since updating the installation to v4.220 seems to have resolved the issue on Havarti

closed 2022-04-13 by EclecticGriffin