calyxir/calyx #989: [Cider] Inconsitencies with watch/breakpoints on a group which is called multiple times consecutively.

EclecticGriffin, 2022-04-26:
When a breakpoint or watchpoint is set on a group which is called multiple times in succession it accidentally treats it as happening only once but taking X times longer where X is the number of consecutive repetitions. This is arguably a bug with the state transition in Seq as it does not idle in-between groups so Cider does not detect that the group stopped and then immediately restarted.

sampsyo, 2022-04-26:
Interesting! I know this is a big topic, but I wouldn't quite go so far as to say that `seq` is misbehaving here per se, since we kinda like the flexibility in the semantics of allowing different low-level timing behaviors to all be valid implementations of `seq`. But nonetheless, we would like the debugger to know that there is a transition going on. Maybe it just wants to observe the state at some point "between" the steps in a `seq`… not sure that even remotely makes sense! 🤪

EclecticGriffin, 2022-04-26:
Yeah, it would need to observe the transitional point between the end of the group and its next beginning, which is kinda weird

EclecticGriffin, 2023-05-01:
Gonna close this one for the moment since I don't have plans to fix this in the immediate future and it's unclear if this is an actually useful feature to have.

closed 2023-05-01 by EclecticGriffin