calyxir/calyx #990: Support randomized execution of `par` blocks

rachitnigam, 2022-04-27:
The current implementation of `par` interpretation algorithm always runs the parallel blocks in serial order. We've discussed a simple extension where the order in which the children are run is selected randomly. Following in the footsteps of [the Verona Project][verona-par], we can go one step further and introduce randomized parallel execution as well.

However, since the interpreter enforces disjointness of reads/writes between parallel children, it is not clear what kind of testing this will enable. Maybe the current goal can be to find bugs in the implementation of the interpreter? In the presence of a future synchronizing primitive, this might be useful for discovering deadlocks possibly?

The place to start would be to look at the [`ParInterpreter`](https://github.com/cucapra/calyx/blob/master/interp/src/interpreter/control_interpreter.rs#L442) implementation and changing it.

[verona-par]: https://github.com/microsoft/verona/blob/master/docs/explore.md#systematic-testing

sampsyo, 2022-04-27:
Interesting idea! Linking to some background on the overall design space for `par`: #921 & #932

The way I see it, randomization of parallel interleaving is a "cheap" alternative to soundly detecting/ruling out all nondeterminism. In other words, the "ideal" thing we would want (if it were even possible!) would be to flag an error whenever the program would do something that would make it nondeterministic—i.e., when the output depends on the relative "running speeds" of different `par` arms. But that can be hard to enforce. So randomization is a cheap/easy way to expose nondeterminism that _does_ exist. A good reference about this stuff is [Chess](https://www.microsoft.com/en-us/research/wp-content/uploads/2016/02/tr-2007-149.pdf), which does a more systematic thing to expose concurrency bugs instead of just randomizing.

I think it's super important to think about the _sources_ of nondeterminism this is supposed to debug. Namely:

- Conflicting writes from different `par` arms to the same port? This is the sort of thing that DRF0 is meant to rule out, and it's the thing that Cider already catches in some circumstances (although not read/write races of the same flavor). So it's not clear what randomization adds on top of the dynamic race detection we're already doing.
- Nondeterminism due to different orderings of updates to state? That is, there could be no data race on any port, but somehow different `par` arms succeed in updating and reading some sequential state (e.g., the contents of a memory). It's not 100% clear how this could happen, but randomization could be effective in uncovering this nondeterminism, which is undesirable _even though it does not violate the semantics of the language_.
- Like the previous bullet point, but in the presence of proper synchronization (i.e., #865)? So again, the programs in question are valid according to some `par` semantics that has yet to be fully written down, but we are worried about nondeterminism in state updates. Randomization could certainly help, but it would need to be done in the context of interpreter support for synchronization. That means that running every `par` branch to completion, in some arbitrary sequence, does not suffice—we need a way to "pause" threads when they synchronize and then "resume" them when they're ready. Then the randomization would apply to the order in which we wake up these threads. But all this is predicated on having a v1.0 support for synchronization that schedules threads in _one_ valid order.

EclecticGriffin, 2022-04-27:
minor wording quibbling 
> The current implementation of par interpretation algorithm always runs the parallel blocks in serial order. 

I would describe the current par interpretation as an interleaved serial execution. In the order they are given each arm gets to do a single step per step of the par block. Because of limitations in the environment, shared cell state can observe this meaning the arms commit state changes in the order defined, but their executions are still concurrent. This is different than interpreting each arm fully before moving to the next.

Anyway, that aside, I think from a technical standpoint this isn't particularly hard to implement as it just requires a little extra machinery and some modifications to the ParInterpreter's `step` method, but I am not entirely sure how to reason åbout what this would get us. If we ultimately correct the limitations of the environment---this being a necessary precursor to implementing synchronization primitives---then each arm will run truly independently and so absent hitherto undefined parallelism rules, there wouldn't be anything gained. If we don't change the environment, then this could help expose certain  read in one thread write in another race conditions under an interleaved interpretation with the ordering being randomized which could be useful, but is really just highlighting the lack of conditions under DRF0. Did any of that make sense?

I should probably open an issue somewhere outlining the environment changes required

rachitnigam, 2022-04-27:
> I should probably open an issue somewhere outlining the environment changes required

This seems like the place to do it!

EclecticGriffin, 2023-05-01:
Closing this for the moment, will revisit later on in the redesign efforts

closed 2023-05-01 by EclecticGriffin