calyxir/calyx #1001: [CIRCT] Lowering high level MLIR programs to Calyx.

cgyurgyik, 2022-05-07:
This documents progress on lowering a high level program to native* Calyx and then through Cider. The high-level conversion path is nicely illustrated by the following infographic [[1]]: 

[image]

PyTorch -> Linalg -> Affine -> StaticLogic (CIRCT) -> Calyx (CIRCT) -> Calyx (Native).

\* _"Native"_ Calyx is the intermediate representation as implemented in this repository [[2]]. The other representation is the [Calyx dialect](https://circt.llvm.org/docs/Dialects/Calyx/), which follows MLIR standards. There exists emitters to go between the native form and the dialect.

## Requirements:

1. [Install Torch-MLIR](https://github.com/llvm/torch-mlir#install-torch-mlir-snapshot)
2. [Install CIRCT](https://capra.cs.cornell.edu/latte22/paper/2.pdf) (which in turn requires MLIR)
3. [Install Calyx](https://docs.calyxir.org/), [fud](https://docs.calyxir.org/fud/index.html), [Cider](https://docs.calyxir.org/debug/cider.html)

## Lowering

1. PyTorch -> Linalg
```
torch-mlir-opt <%s -convert-torch-to-linalg
```

2. Linalg -> Affine

```bash
# TODO(cgyurgyik): Verify with PyTorch community this is still the bufferization method used.
mlir-opt %s --linalg-comprehensive-module-bufferize='allow-return-memref use-alloca' --convert-linalg-to-affine-loops
```

3. StaticLogic -> Calyx (CIRCT)
```
circt-opt %s --lower-scf-to-calyx='cider-source-location-metadata' -canonicalize
```

4. Calyx (CIRCT) -> Calyx (Native)
```
 circt-translate %s --export-calyx
```

5. Cider
```
fud e --to debugger %s
```

[1]: https://capra.cs.cornell.edu/latte22/paper/2.pdf
[2]: https://arxiv.org/abs/2102.09713

mikeurbach, 2022-05-10:
> TODO(cgyurgyik): Verify with PyTorch community this is still the bufferization method used.

FYI this is definitely not the bufferization method they use, this is a newer method I wanted to try out in hlstool. The hlstool command you've pasted here is actually out of date already by the looks of it. If you are running into bufferization issues, I can try to help use the method torch-mlir uses, or adopt the latest and greatest.

cgyurgyik, 2022-05-10:
I must have misunderstood our chat offline. At this point, I'm just trying to lower something that the StaticLogic will nicely digest.

cgyurgyik, 2022-05-25:
So I've been working on lowering Affine -> Calyx native. 

First, a couple bugs with a [hacky fix](https://github.com/llvm/circt/pull/3130): 
- https://github.com/llvm/circt/issues/3111
- https://github.com/llvm/circt/issues/3112

Trying to run anything a bit more complex than the tests provided in the CIRCT suite has led to compiler crashes, and more recently, bits being removed by `std_pad` which wasn't caught by the interpreter because this is sometimes expected behavior. Instead of trying to introduce more brittle fixes, I've decided the next best step for longevity is to finally separate the two passes: https://github.com/llvm/circt/issues/2988 (tracker). @mortbopet has also suggested the following clean-up for the StaticLogic side: https://github.com/llvm/circt/issues/3113. However, we still need to finish the separation clean-up (WIP).

Part of this issue is placing all the necessary steps in one location. Yes, StaticLogic -> Calyx native is possible. It just isn't very robust currently.

cgyurgyik, 2022-05-25:
> > Part of this issue is placing all the necessary steps in one location.
> 
> For the record, this flow does exist in one place, wrapped up by a Python script here: https://github.com/circt-hls/circt-hls/tree/main/tools/hlstool.

I haven't used hlstool, but it definitely looks like something good to have in the workflow. My goal here was to provide minimal commands.

> I'd be curious to take a look if you have more examples of failures, the pass should emit an error instead of crashing.

Yeah I can bring up more examples, but honestly it wasn't hard for me to cause the compiler to crash / produce semantically incorrect code (e.g., slicing bits that shouldn't be sliced) - this is without any loop nesting. After the initial hacky fix, it seemed like the best next step was to separate SCF and StaticLogic code as an intermediary step. 

Then, we can re-produce these bugs and hopefully write better fixes for them that don't require reasoning about two conversions. Curiously, do you have examples of bigger programs you've successfully lowered to Calyx native with desired behavior?

mikeurbach, 2022-05-25:
> Curiously, do you have examples of bigger programs you've successfully lowered to Calyx native with desired behavior?

Nothing more complex than a single loop nest, like what is checked into CIRCT's tests.

mikeurbach, 2022-05-25:
> Yeah I can bring up more examples, but honestly it wasn't hard for me to cause the compiler to crash / produce semantically incorrect code (e.g., slicing bits that shouldn't be sliced) - this is without any loop nesting.

Please file issues on CIRCT. No one other than myself has tested this flow, and I've only tested the simplest meaningful example I could write: a dot product. This is very experimental work in progress.

rachitnigam, 2022-11-13:
@cgyurgyik are there action items for the this tracker that we should be working on? If not, let's wrap up the issue

closed 2023-04-29 by rachitnigam