calyxir/calyx #1004: Running Forever on Icarus Verilog Backend

dc854, 2022-05-13:
I compiled the following Dahlia code [depth.fuse](https://github.com/cucapra/dahlia/files/8690993/depth.txt) into Calyx  [depth.futil](https://github.com/cucapra/dahlia/files/8690995/depth-calyx.txt) and ran into the issue where Icarus Verilog would run forever, but it would finish running immediately in Verilator.

This is the data file I used [depth.futil.data](https://github.com/cucapra/dahlia/files/8691002/depth-data.txt), and I used `fud e --from dahlia depth.fuse --to futil` to compile it to Calyx, then used
```
fud exec depth.futil --to dat --through icarus-verilog \
    -s verilog.data depth.futil.data
```
to run it through Icarus Verilog.

cgyurgyik, 2022-05-14:
```
decl paths: ubit<32>[20];
decl offsets: ubit<32>[3];
decl segments: ubit<32>[15];

let j : ubit<32>= 0;
let max : ubit<32>= 0;
for (let i: ubit<2> = 0..3) {
  max := offsets[i+1];
  while (j < max) {
      let index = paths[j];
      j := j + 1;
      let count = segments[index];
      ---
      segments[index] := count + 1;
  }
}
```

```
  "offsets0": {
    "data": [
      0,
      10,
      20
    ],
    "format": {
      "numeric_type": "bitnum",
      "is_signed": false,
      "width": 32
    }
  }
```

The length of `offsets0` is 3. Index `i` may be any value in the range [0, 2]. Let `j := i + 1`. Then index `j` is any value in the range [1, 3]. You have an out-of-bounds access with the following LOC:

```
  max := offsets[i+1];
```

EclecticGriffin, 2022-05-14:
The interpreter corroborates this as well 

```
$ fud e --to interpreter-out depth.fuse 
[fud] ERROR: `/Users/griffin/research/capra/calyx/target/debug/interp   -l /Users/griffin/research/capra/calyx   /var/folders/4s/d76r5g0j6pg3xgq7hnhkng200000gn/T/tmp1ofdlq0v' failed:
=====STDERR=====
Error: invalid memory access to memory main.offsets0. Given index (3) but memory has dimension (3)

=====STDOUT=====

```

EclecticGriffin, 2022-05-14:
A thought for @rachitnigam, is it possible to set an assertion for verilator that checks this for the memory primitives? In general this UB is a real foot-gun especially given the difference between simulators

EclecticGriffin, 2022-05-14:
I suppose the secret (and probably bad) third option would be to enforce that an invalid index to the memory primitives always outputs zero, but I feel like that's the wrong way to go even if it would let icarus and verilator agree

rachitnigam, 2022-05-14:
We could probably do something like that but in the expected case, where the `idx_size` is `log(size)`, this would not be detectable

rachitnigam, 2022-05-14:
@dc854 does fixing the out of bound access fix the program for you?

sampsyo, 2022-05-16:
Indeed, great that the interpreter catches this! While catching this in Verilog simulators would be _somewhat_ nice, I kinda feel comfortable with this being an interpreter-exclusive enforcement mechanism too. Maybe we should explicitly document a debugging workflow that looks sort of like this:

1. Try it in Verilator/Icarus initially.
2. If something goes wrong there, run the same thing in the interpreter (easy peasy: just switch your `fud` flags to `--to interpreter-out`).
3. If something _still_ seems wrong, file a bug. 😄

Not exactly sure where this would go in the docs though… maybe a new page in the [tutorial section](https://docs.calyxir.org/tutorial/language-tut.html)?

dc854, 2022-05-18:
Seems like the array bounds was the issue. I did try it out on the interpreter before, but something I was doing caused it to not run into that error. I just tried stepping through the entire program, and it does seem to fail as expected though. Thanks for the help!

PR #1012 (2022-05-26): Add out of bounds checking to memory primitives
Add out of bounds checking to memory primitives for the verilator backend. Fixes #1004.
closed 2022-05-26 by rachitnigam