calyxir/calyx #1020: Get dot product executing on FPGA

nathanielnrn, 2022-06-07:
1st part of #1019 

Goal is to become familiar with steps required to manually run Verilog on FPGA boards in order to generalize designs' execution in the future.

Executing on FPGAs can be broken down as follows:

- [ ] Set up remote (Gorgonzola) and local Xilinx toolchain, namely, Vivado, Vivado HLS and Vitis
- [ ] Perform Verilog simulation (hw_emu) using Xilinx Tools, see [here](https://docs.calyxir.org/fud/xilinx.html) 
- [ ] Manually write an AXI interface. See [here](https://support.xilinx.com/s/article/1053914?language=en_US)
- [ ] Map and place design on FPGA using existing place & route implementation
- [ ] Write XRT host code to run the FPGA design
- [ ] Verify outputs

Additionally, this issue will be home to files written, low-level questions, and miscellaneous auxiliary files and data.

nathanielnrn, 2022-06-10:
So it seems like 

>  Manually write an AXI interface. See [here](https://support.xilinx.com/s/article/1053914?language=en_US)

May not be the best approach to tackling this issue (although I'm happy to hear more opinions on this).
See #1022 June 10 comment

rachitnigam, 2022-07-29:
@nathanielnrn this works now right?

nathanielnrn, 2022-07-29:
Yes, see #1072 for details

closed 2022-07-29 by nathanielnrn