calyxir/calyx #1028: [CIDR] Debugger does not work after violated bounds error message.

cgyurgyik, 2022-06-09:
```bash
fud e mlir.futil --to debugger -s verilog.data mlir.data -q
== Calyx Interactive Debugger ==
 > continue
Jun 09 03:42:00.030 WARN While loop has violated its bounds. The annotation suggests that the body should execute 3 times, but it exited after 2 iterations. 
     147 |        @bound(3) while std_lt_0.out with bb0_0 {
, source: main
```

After running into a violated bounds attribute in https://github.com/llvm/circt/issues/3311, the debugger just hangs up, i.e., you cannot type anymore commands. It's not obvious to me why this is occurring. Trying to minimize this works as expected, e.g.,

```
import "primitives/core.futil";
import "primitives/binary_operators.futil";
component main<"toplevel"=1>(ext_mem0_read_data: 32, ext_mem0_done: 1, ext_mem1_read_data: 32, ext_mem1_done: 1, ext_mem2_read_data: 32, ext_mem2_done: 1, @clk clk: 1, @reset reset: 1, @go go: 1) -> (ext_mem0_write_data: 32, ext_mem0_addr0: 2, ext_mem0_write_en: 1, ext_mem1_write_data: 32, ext_mem1_addr0: 2, ext_mem1_write_en: 1, ext_mem2_write_data: 32, ext_mem2_addr0: 1, ext_mem2_write_en: 1, @done done: 1) {
  cells {
    std_add_0 = std_add(32);
    std_lt_0 = std_lt(32);
    while_0_arg0_reg = std_reg(32);
  }
  wires {
    group assign_while_0_init_0 {
      while_0_arg0_reg.in = 32'd0;
      while_0_arg0_reg.write_en = 1'd1;
      assign_while_0_init_0[done] = while_0_arg0_reg.done;
    }
    comb group bb0_0 {
      std_lt_0.left = while_0_arg0_reg.out;
      std_lt_0.right = 32'd4;
    }
    group bb0_3 {
      std_add_0.left = while_0_arg0_reg.out;
      std_add_0.right = 32'd1;
      while_0_arg0_reg.in = std_add_0.out;
      while_0_arg0_reg.write_en = 1'd1;
      bb0_3[done] = while_0_arg0_reg.done;
    }
  }
  control {
    seq {
      assign_while_0_init_0;
      @bound(1) while std_lt_0.out with bb0_0 { seq { bb0_3; } }
    }
  }
}
```

```bash
fud e mlir.futil --to debugger -s verilog.data mlir.data -q
== Calyx Interactive Debugger ==
 > c
Jun 09 03:48:45.981 WARN While loop has violated its bounds. The annotation suggests that the body should execute 1 times, but it has entered its 2 iteration. 
     82 |      @bound while std_lt_0.out with bb0_0 {
, source: main
Jun 09 03:48:45.982 WARN While loop has violated its bounds. The annotation suggests that the body should execute 1 times, but it has entered its 3 iteration. 
     82 |      @bound while std_lt_0.out with bb0_0 {
, source: main
Jun 09 03:48:45.983 WARN While loop has violated its bounds. The annotation suggests that the body should execute 1 times, but it has entered its 4 iteration. 
     82 |      @bound while std_lt_0.out with bb0_0 {
, source: main
Main component has finished executing. Debugger is now in inspection mode.
Jun 09 03:48:45.984 WARN While loop has violated its bounds. The annotation suggests that the body should execute 1 times, but it exited after 4 iterations. 
     82 |      @bound while std_lt_0.out with bb0_0 {
, source: main
>
```

cgyurgyik, 2022-06-09:
Fixing the bounds check in the bigger example and just trying to run it leads to a hang-up, so unsure what the root cause is.

cgyurgyik, 2022-06-09:
Ah, the hang-up is probably due to an external signal that isn't being driven. I'll re-open this if that's not the case.

closed 2022-06-09 by cgyurgyik

sampsyo, 2022-06-09:
Right; a hopefully useful alternative to solving the halting problem is to just detect when we've been waiting for "a while" and show where. Manually interrupting execution is one valid implementation of "a while."