calyxir/calyx #1031: Calyx Documentation is Missing Information on '?' Symbol

susan-garry, 2022-06-13:
The language tutorial for calyx, found [here](https://docs.calyxir.org/tutorial/language-tut.html), does not include any information about the `?` and `&` symbols. These are key features of the language, but I haven't been able to find any information about it in the documentation (they are used in examples under [debugging tips](https://docs.calyxir.org/debug/debug.html), but their semantics are not explained). Information about these keywords should probably be added to the tutorial.

rachitnigam, 2022-06-13:
Ah, good point! We should probably explain this somewhere in detail.

sampsyo, 2022-06-14:
Yes indeed! I think it would be cool to add this in another stage in the language tutorial… it could be kinda contrived, but we could just add a phase where the loop only adds up odd numbers, skipping the even ones, or something like that.

closed 2022-11-28 by rachitnigam