calyxir/calyx #1040: `futil` not found when running outside of `calyx/` dir

gussmith23, 2022-06-16:
I've built Calyx inside a Docker image with the following:

```docker
RUN cargo build --manifest-path ./calyx/Cargo.toml \
  && cargo install runt \
  && FLIT_ROOT_INSTALL=1 flit -f ./calyx/fud/pyproject.toml install -s --deps production \
  && mkdir /root/.config \
  && fud config global.futil_directory /root/calyx
```

When I then run:
```sh
docker run -it my-image bash -c 'fud config global.futil_directory /root/calyx && fud e calyx/examples/tutorial/language-tutorial-iterate.futil \
  -s verilog.data calyx/examples/tutorial/data.json \
  --to dat --through verilog -v'
```
I get:
```
✖ futil.run_futil
[fud] ERROR: `./target/debug/futil -l /root/calyx -b verilog' failed:
=====STDERR=====
/bin/sh: 1: ./target/debug/futil: not found

=====STDOUT=====
```

Things work if I `cd` first, though:
```sh
docker run -it my-image bash -c 'cd calyx && fud config global.futil_directory /root/calyx && fud e examples/tutorial/language-tutorial-iterate.futil \ 
  -s verilog.data examples/tutorial/data.json \ 
  --to dat --through verilog -v'
✔ futil.run_futil
✔ transform
✔ verilog.mktmp
✔ verilog.json_to_dat
✔ verilog.compile_with_verilator
✔ verilog.simulate
✔ verilog.output_json
✔ verilog.cleanup
{
  "cycles": 85,
  "memories": {
    "mem": [
      42
    ]
  }
}```

gussmith23, 2022-06-16:
Does `fud` have to use a relative directory for `futil`? I would expect to be able to use `fud` from anywhere.

sampsyo, 2022-06-16:
Hmm; that's odd! My best guess is that your `cargo build` command is building the library only, not the executable. There's a `Cargo.toml` at the root that is for the compiler executable, as well as in the `calyx` subdirectory (which it currently looks like you're building) that is just for the core library.

Barring that, can you do us a quick favor and run a plain ol `fud config` inside the container to see if that `fud config global.futil_directory /root/calyx` command took hold correctly?

sampsyo, 2022-06-16:
Actually, that doesn't explain why it would work when you `cd` to the right place. So my best guess is now the configuration thing…

gussmith23, 2022-06-16:
```

[global]
futil_directory = "/root/calyx"

[stages.futil]
exec = "./target/debug/futil"
file_extensions = [ ".futil",]

[stages.interpreter]
exec = "./target/debug/interp"
round_float_to_fixed = true

[stages.debugger]

[stages.dahlia]
exec = "dahlia"
file_extensions = [ ".fuse", ".dahlia",]

[stages.verilog]
exec = "verilator"
file_extensions = [ ".v", ".sv",]
cycle_limit = 500000000
top_module = "main"
round_float_to_fixed = true

[stages.vcd]
exec = "vcdump"
file_extensions = [ ".vcd",]

[stages.vcd_json]
file_extensions = [ ".json",]

[stages.jq]
exec = "jq"

[stages.dat]
file_extensions = [ ".dat",]

[stages.relay]
file_extensions = [ ".relay",]

[stages.systolic]
file_extensions = [ ".systolic",]

[stages.synth-verilog]
exec = "vivado"
ssh_host = ""
ssh_username = ""

[stages.vivado-hls]
exec = "vivado_hls"
ssh_host = ""
ssh_username = ""

[stages.xclbin]
mode = "hw_emu"
device = "xilinx_u50_gen3x16_xdma_201920_3"
temp_location = "/tmp"
ssh_host = ""
ssh_username = ""

[stages.wdb]
mode = "hw_emu"
ssh_host = ""
ssh_username = ""
xilinx_location = "/scratch/opt/Xilinx/Vitis/2020.2"
xrt_location = "/opt/xilinx/xrt"

[stages.fpga]
```

thanks for the quick reply adrian :)

sampsyo, 2022-06-16:
Cool; thanks! Maybe give this a shot?

```
fud config stages.futil.exec /root/calyx/target/debug/futil
```

(I'm ever so slightly confused because I vaguely remember this working without fixing a path for this separately from the root directory, but I don't now see how that could have been possible, and it's possible I hallucinated that. Anyway, seems like something we could make a little easier somehow?)

rachitnigam, 2022-06-16:
Nope, you need to provide the full path to fud before it starts working outside of the calyx directory. “Fud check” will warn you about this problem. The fix is using the “fud c” command Adrian mentioned

rachitnigam, 2022-06-16:
The futil_directory stuff is used for different things and is not used to make calls to compiler work globally

gussmith23, 2022-06-16:
Works! Thanks!

closed 2022-06-16 by gussmith23