calyxir/calyx #1041: Bug in Continuous Assignments in Live Range Analysis

calebmkim, 2022-06-17:
Say we have a Calyx program like this: 
```
component main(@go go: 1, @clk clk: 1, @reset reset: 1) -> (@done done: 1) {
  cells {
    r0 = std_reg(32);
    r1 = std_reg(32);
    add0 = std_add(32);
    add1 = std_add(32);
  }
  wires {
    add0.left = r1.out;
    add1.left = r0.out;
    done = r0.done;
  }

  control { 
  }
}
```
Running just the `minimize-regs` pass will share registers, like this: 
```
add0.left = r0.out;
add1.left = r0.out;
done = r0.done;
```
Talking w/ @sampsyo we concluded that this was incorrect behavior. 
We think it has to do with the following code, found in live_range_analysis.rs line 251. This code occurs after the data flow stuff for the live range analysis has been done, and is trying to deal with cells that are used in continuous assignments. 
```
// add global reads to every point
let global_reads: Prop =
    ReadWriteSet::read_set(comp.continuous_assignments.iter())
        .filter(|c| c.borrow().type_name() == Some(&"std_reg".into()))
        .map(|c| c.clone_name())
        .collect::<HashSet<_>>()
        .into();
for (_, prop) in ranges.live.iter_mut() {
    *prop = &*prop | &global_reads;
}
```
Two things: 
1) It takes all registers that are read from in continuous assignments, and assigns them as "live" in every group. But the original Calyx program we have doesn't have any groups, so it doesn't create a conflict between these two registers being read from; hence the sharing. If we add a dummy group to the program that does basically nothing, the registers are no longer shared. This is an important edge case to keep in mind for future live-range-analysis testing. 
2) Why do we only look at the cells that are *read* from continuous assignments? My idea currently is that we should never share cells that are used (i.e. read or written) in any continuous assignments?  In other words, the way I am writing the combined resource-sharing pass, if any cells that are read or written from in a continuous assignment, they are eliminated from the possibility of being shared. @rachitnigam lmk if I'm missing something here.

rachitnigam, 2022-06-22:
Yup, this looks like a bug and your intuition seems correct–a cell that is *used* in continuous assignments should be considered unshareable

rachitnigam, 2022-06-29:
@calebmkim was this fixed by you generalize sharing pass?

calebmkim, 2022-06-29:
Yes I will close it

closed 2022-06-29 by calebmkim