calyxir/calyx #1048: Eliminate pointless writes

rachitnigam, 2022-06-21:
Software compilers can optimize the following code:
```
x = 1
y = x
z = y + 1
```
To:
```
x = 1
z = x + 1
```

The goal of building such a pass in Calyx is computing writes to registers that are not needed. A common place where this occurs is saving the value generated by a port marked as `@stable` into a register. For example, the `out` port on a `mult_pipe` is `@stable` and therefore can be read directly:
```
group do_mult {
  mult.go = !mult.done ? 1;
  mult.left = ...; mult.right = ...;
  r.in = mult.done ? mult.out;
  r.write_en = mult.done ? 1;
  do_mult[done] = r.done;
}
group use_r {
  ...
  add.left = r.out;
}
```

This group saves the value computed by the `mult` into the register `r`. However, `mult.out` is going to be available till the next invocation of `mult` which means we can just use the value from `mult.out` directly:
```
group do_mult {
  mult.go = 1;
  mult.left = ...; mult.right = ...;
  do_mult[done] = mult.done;
}
group use_r {
  ...
  add.left = mult.out;
}
```

Note that optimization pointless writes to a register is the same problem because register ports are marked `@stable` as well.
Similarly, this pass should also eliminate groups that perform writes to a register that is never used in the future again.

rachitnigam, 2022-06-22:
There is some description of what `@stable` does in the documentation: https://docs.calyxir.org/lang/attributes.html#stable

sampsyo, 2022-06-22:
Ah, thanks! My grep for `@stable` failed me because it doesn't have the sigil. 🤣 

Anyway, maybe we need to nail it down in a bit more detail to do this work… TBH, I'm still a teensy bit fuzzy about what it's supposed to guarantee. Yes, it stays "live" beyond a single group, but for how long? Not indefinitely, to be sure…

rachitnigam, 2022-06-25:
Yeah, I think it better not be something that changes over time. I think "stable" should mean something like: The port retains the *exact* same value till the next invocation (or use) of the component. A port can only be stable if it is connected to another stable port *and* that port does not get updated between two invocations.