calyxir/calyx #1060: Still getting errors from `--to resource-estimate`

gussmith23, 2022-06-29:
output:

```
RUST_BACKTRACE=full fud e --to resource-estimate examples/futil/dot-product.futil 
[fud] ERROR: `cd /tmp/tmpv2uomtei && vivado -mode batch -source synth.tcl>&2' failed:
=====STDERR=====

****** Vivado v2021.2 (64-bit)
  **** SW Build 3367213 on Tue Oct 19 02:47:39 MDT 2021
  **** IP Build 3369179 on Thu Oct 21 08:25:16 MDT 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source synth.tcl
# set outdir ./out
# set partname "xczu3eg-sbva484-1-e"
# create_project -force -part $partname FutilBuild $outdir
# add_files [glob ./*.sv]
# add_files -fileset constrs_1 [glob ./*.xdc]
# set_property top main [current_fileset]
# set_property \
#     -name {STEPS.SYNTH_DESIGN.ARGS.MORE OPTIONS} \
#     -value {-mode out_of_context -flatten_hierarchy "rebuilt"} \
#     -objects [get_runs synth_1]
# launch_runs synth_1
WARNING: [Vivado 12-7122] Auto Incremental Compile:: No reference checkpoint was found in run synth_1. Auto-incremental flow will not be run, the standard flow will be run instead.
[Wed Jun 29 02:55:56 2022] Launched synth_1...
Run output will be captured here: /tmp/tmpv2uomtei/out/FutilBuild.runs/synth_1/runme.log
# wait_on_run synth_1
[Wed Jun 29 02:55:56 2022] Waiting for synth_1 to finish...

*** Running vivado
    with args -log main.vds -m64 -product Vivado -mode batch -messageDb vivado.pb -notrace -source main.tcl

****** Vivado v2021.2 (64-bit)
  **** SW Build 3367213 on Tue Oct 19 02:47:39 MDT 2021
  **** IP Build 3369179 on Thu Oct 21 08:25:16 MDT 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source main.tcl -notrace
Command: synth_design -top main -part xczu3eg-sbva484-1-e -mode out_of_context -flatten_hierarchy rebuilt
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xczu3eg'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xczu3eg'
INFO: [Device 21-403] Loading part xczu3eg-sbva484-1-e
INFO: [Synth 8-7079] Multithreading enabled for synth_design using a maximum of 4 processes.
INFO: [Synth 8-7078] Launching helper process for spawning children vivado processes
INFO: [Synth 8-7075] Helper process launched with PID 85265
ERROR: [Synth 8-6735] net type must be explicitly specified for 'in' when default_nettype is none [/tmp/tmpv2uomtei/main.sv:753]
INFO: [Synth 8-2350] module std_wire ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:750]
INFO: [Synth 8-2350] module std_slice ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:759]
INFO: [Synth 8-2350] module std_pad ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:780]
INFO: [Synth 8-2350] module std_not ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:802]
INFO: [Synth 8-2350] module std_and ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:811]
INFO: [Synth 8-2350] module std_or ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:821]
INFO: [Synth 8-2350] module std_xor ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:831]
INFO: [Synth 8-2350] module std_add ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:841]
INFO: [Synth 8-2350] module std_sub ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:851]
INFO: [Synth 8-2350] module std_gt ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:861]
INFO: [Synth 8-2350] module std_lt ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:871]
INFO: [Synth 8-2350] module std_eq ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:881]
INFO: [Synth 8-2350] module std_neq ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:891]
INFO: [Synth 8-2350] module std_ge ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:901]
INFO: [Synth 8-2350] module std_le ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:911]
INFO: [Synth 8-2350] module std_lsh ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:921]
INFO: [Synth 8-2350] module std_rsh ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:931]
INFO: [Synth 8-2350] module std_mux ignored due to previous errors [/tmp/tmpv2uomtei/main.sv:943]
Failed to read verilog '/tmp/tmpv2uomtei/main.sv'
INFO: [Common 17-83] Releasing license: Synthesis
24 Infos, 0 Warnings, 0 Critical Warnings and 2 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Wed Jun 29 02:56:16 2022...
[Wed Jun 29 02:56:27 2022] synth_1 finished
WARNING: [Vivado 12-8222] Failed run(s) : 'synth_1'
wait_on_runs: Time (s): cpu = 00:00:25 ; elapsed = 00:00:31 . Memory (MB): peak = 2597.195 ; gain = 0.000 ; free physical = 51225 ; free virtual = 70228
# launch_runs impl_1 -to_step route_design
ERROR: [Common 17-70] Application Exception: Failed to launch run 'impl_1' due to failures in the following run(s):
synth_1
These failed run(s) need to be reset prior to launching 'impl_1' again.

INFO: [Common 17-206] Exiting Vivado at Wed Jun 29 02:56:27 2022...

=====STDOUT=====

```

Any ideas? Seems like it's about `default_nettype`.

rachitnigam, 2022-06-29:
Looks like the std library in primitives/core has a bunch of modules where the input and/or output ports don’t have the logic keyword in front of them. Maybe try adding them and see if that works?

rachitnigam, 2022-06-29:
Trying adding wire keyword in this file here: https://github.com/cucapra/calyx/blob/master/primitives/core.sv#L23

gussmith23, 2022-06-29:
Yup, works! thank you!