calyxir/calyx #1061: Building a barrier from `std_sync_reg`

paili0628, 2022-06-29:
We are building a `std_sync_reg` with multiple readers and writers. The register blocks all writes when it has a value, which means when it is "full". It also blocks all reads when it does not have a value, i.e., is "empty". When multiple writers try to write to the register, it only chooses one value from the writers to write in, and blocks all other writes. When multiple readers try to read from the register, it only gives its value to one reader; all other readers get blocked. After the value is read, it is destroyed and the register is again set to "empty".

We would like to implement the @sync syntax (discussed in #932) with this primitive. In other words, we would like to implement a barrier: a structure to ensure that threads running in parallel all get to a certain point of the program before they proceed together. How to do this using `std_sync_reg`? We can think of it as a counter. When the program starts, the counter is automatically set to 0. When each thread gets to the stopping point, i.e., the point where it is told by the program to stop and wait for all other threads to finish, it tries to read from the register. When it eventually gets a value from the register, it increments it and writes the incremented value back in. Because all operations can be thought of as atomic(since the value is immediately destroyed when it is read by one reader), we can guarantee that the counter is accurate. 

How do we signify to all threads that it is now safe to proceed? One solution is to have a 'peek' value, which is a wire that holds the same value as the register. It gets updated every time the value in the register is updated but is never destroyed. Threads are notified  to proceed when the 'peek' value reaches the total number of threads of the program. This way threads can proceed immediately when possible.

Created through discussion with @rachitnigam and @sampsyo.

sampsyo, 2022-06-29:
Nice summary, @paili0628; thanks!! One detail I wanted to add:

In the idea we discussed yesterday, the barrier actually consists of *two* `std_sync_reg` cells. The first contains the counter you described—it essentially implements an atomic incremented. Then we have a *second* `std_sync_reg` that is just for "waiting." It remains "empty" as long as the barrier has not reached capacity and, when the last thread finally arrives, becomes "full" only then.

rachitnigam, 2022-06-30:
I think @paili0628's write up is about the alternative design where we have access to a `peek` method that all threads to unblock as soon as the value in the cell becomes == number of threads

paili0628, 2022-06-30:
Yes, the alternative design avoids this problem altogether since theoretically the threads would be notified to proceed at the exact same time.

closed 2022-08-03 by paili0628