calyxir/calyx #1071: Issues with the AXI implementation

andrewb1999, 2022-07-01:
I spent some time today getting the dot-product design working with AXI on real FPGAs. I just manually edited the verilog for now, but I'll document the issues I found so they can be fixed in the generator at some point. Here's the repo with the working design: https://github.com/andrewb1999/calyx-axi-ex.git

### Actual Calyx issues (dot-product.sv)
It's a bad idea to initialize logic (especially resets) that are used as wires rather than registers. I just commented out the entire initial block in the verilog which prevented issues with std_regs not being reset properly (currently none of the logic defined in the main module is being used as a register so this is fine to do).

### XML issues (kernel.xml)
The datawidth of the m axi ports need to be 512 because this is the width we actually use in hardware.

### Wrapper (wrapper.v)

Don't have time to finish listing these tonight (this is where the core bug fixes are). I'll post this issue anyways for now and finish typing these out tomorrow.

andrewb1999, 2022-07-02:
@nathanielnrn @sampsyo @rachitnigam I updated this issue with all the bugs I found and explanations. Let me know if you have any questions, I have a pretty good grasp on what's going on now so I should be able to help explain stuff.

rachitnigam, 2022-07-02:
Thank you for the incredible debugging work @andrewb1999!! This will be really useful in getting a push button FPGA flows for calyx stuff!

nathanielnrn, 2022-07-05:
This is amazing! I'm implementing changes to code generation now.

Sorry if this question has an obvious answer, I looked around but couldn't find one. Thought it might save time asking here.

Where exactly does the kernel.xml interface with verilog code? In other words, would changes need to be made to any port widths in verilog with respect to 

> The datawidth of the m axi ports need to be 512 because this is the width we actually use in hardware.

andrewb1999, 2022-07-05:
@nathanielnrn The kernel.xml doesn't really "interface" with the verilog per-say, it is just a simpler description of the underlying verilog that XRT uses to know how to interact with the kernel. What's important here is that the data width of the RDATA and WDATA channels match the datawidth that we specify in the kernel.xml. So for example, `m0_axi_RDATA` has a datawidth of 512 in the verilog (wire[511:0]), so in the kernel.xml we need to specify that the m0_axi port has a dataWidth of 512. Does that make sense?

sampsyo, 2022-07-06:
Maybe one way of putting it is that the `kernel.xml` file *describes* what's going on in the Verilog.

One thing that's still unclear to me is where the width 512 comes from in the generator code. This is where the data width seems to stem from:
https://github.com/cucapra/calyx/blob/943a50ebdb1533dabfc6f82f638b43d5744fe7c0/src/backend/xilinx/toplevel.rs#L46

…and there it's set to 32. So this may require some debugging!

andrewb1999, 2022-07-06:
That's actually the Control_axi instantiation, so 32 is the correct value there. The issue is with the memory module here:

https://github.com/cucapra/calyx/blob/943a50ebdb1533dabfc6f82f638b43d5744fe7c0/src/backend/xilinx/toplevel.rs#L50

There is a 512 there, but there's also a 64 and a 32 (for address width and the data width of the argument). I think there's just a mismatch somewhere of what value needs to go into the kernel.xml.

sampsyo, 2022-07-07:
Aha, thanks for the correction, @andrewb1999!! That would explain it.