calyxir/calyx #1080: Remove initialization of wires

nathanielnrn, 2022-07-06:


rachitnigam, 2022-07-06:
If it's the verilog backend then here: https://github.com/cucapra/calyx/blob/master/src/backend/verilog.rs#L182

The `--disable-init` flag removes the initialize blocks. If you're talking about the axi backend then I'm not sure but it'll be `axi.rs` probably

nathanielnrn, 2022-07-07:
Quick question that came up while working on this: Should

```
module std_wire #(
  parameter WIDTH = 32
) (
  input wire logic [WIDTH - 1:0] in,
  output logic [WIDTH - 1:0] out
);
  assign out = in;
endmodule
```

`in` port be a wire or just a logic (do people say this? Feels more natural to say reg.) @andrewb1999 changes has this as just a logic. See [changed](https://github.com/andrewb1999/calyx-axi-ex/blob/be953743a976f23954ad6d55a8a4793895bee49a/src_default/dot-product.sv#L753) versus [original](https://github.com/andrewb1999/calyx-axi-ex/blob/be953743a976f23954ad6d55a8a4793895bee49a/src/dot-product.sv#L753)

rachitnigam, 2022-07-07:
Thats because we merged a change since then. You can look at the blame to see when that changed but I don't think you need to change anything here

PR #1085 (2022-07-07): Added --disable-init flag to fud xclbin stage
Fixes #1080, which is part of #1072
closed 2022-07-07 by nathanielnrn