calyxir/calyx #1082: Fix order of operations of addr calculations (bit shift must occur before addition)

nathanielnrn, 2022-07-06:


nathanielnrn, 2022-07-06:
Another one I'm not sure about @rachitnigam. The relevant generation occurs [here.](https://github.com/cucapra/calyx/blob/924d879764c669fc464b75f8286e1a989e252c2e/src/backend/xilinx/memory_axi.rs#L204-L206) COuld you point me in the direction of parentheses logic//forcing parentheses?

rachitnigam, 2022-07-08:
@nathanielnrn can you open an issue in the vast library saying what the expected behavior vs. actual behavior is and tag me? I can fix it once I have a test. An easy way to figure this out would be to clone the vast repo and add a new test here: https://github.com/vegaluisjose/vast/blob/ae19379034a2d520cd572c0a1cac2f735f8269a6/tests/v05.rs

The test would let you specify what the output verilog should look like

closed 2022-07-11 by nathanielnrn