calyxir/calyx #1099: Specifying latencies for multi-ported memories

rachitnigam, 2022-07-13:
In order to support multi-ported memories with static latencies, we need some way of specifying the latency of each port. Currently, we specify the latency of a memory using:
```
primitive std_mem_d1<"static"=1>(@go write_en: 1, ..) -> (@done done: 1, ..);
```

Implicitly, the `static` annotation corresponds to the number of cycles between when `go` is asserted to when `done` is asserted. In the case of a mutli-ported memory, however, we need to specify latencies for multiple go-done paths:

```
primitive mem_d1_two_ports<"static_1"=1, "static_2"=2>(
  @go(1) write_en_1: 1,
  @go(2) write_en_2: 1, ..
) -> (
  @done(1) done_1: 1,
  @done(2) done_2: 1, ..
);
```

The way to interpret this interface is that for the `go(n)-done(n)` pair, the latency is specified by the `static_n` attribute. If `static_n` is missing, then this path is dynamic.

## Using Multiple `static` Annotations

The key place to use these annotations is in `infer-static-timing` which is responsible for figuring out a group's latency. Once a group has been marked `@static` by this pass, the `static-timing` pass takes care of compiling it in a latency-sensitive manner.

We can extend the logic in `infer-static-timing` to make it consider `go-done` pairs along with `static` annotations in a pretty straightforward way.

## Method-based Invokes

Because we currently do not support method-based invokes (#986), we cannot apply `group2invoke` on groups that just perform memory reads. In the future, when we add support for it, we can consider applying `group2invoke`.

rachitnigam, 2022-07-13:
CC @andrewb1999

sampsyo, 2022-07-14:
Nice work; very clean extension! Hopefully the changes to `infer-static-timing` aren't too bad.

In the broader "methods" proposal #986, it seems like the static attribute would want to become part of a given method (or designated/named `go`/`done` pair).

rachitnigam, 2022-07-17:
I proposed a small modification of this to @andrewn1999 where instead of specifying the `static` annotations on a component, we specify it on a port marked with `@go`:
```
primitive mem_d1_two_ports(
  @static(1) @go(1) write_en_1: 1,
  @static(2) @go(2) write_en_2: 1, ..
) -> (
  @done(1) done_1: 1,
  @done(2) done_2: 1, ..
);
```

This has two benefits:
1. It reflects the fact that staticness is not a property of a component but instead that of a specific `go`-`done` path through the circuit.
2. We don't need the janky `static_n` stuff express which static attribute corresponds to which `go(n)` signal

sampsyo, 2022-07-18:
Great idea. Seems much nicer.

PR #1125 (2022-07-29): Support multiple `@static` attributes
Fixes #1099. Couple of things that need to be addressed:
- [x] `compile-invoke` should fail if a primitive has more than one `@go`-`@done` interface.
- [x] `group2invoke` needs to be reason about multiple `@go`-`@done` specs being present.
- [x] Add test with multiple specs
- [x] Change the existing primitives to use the new syntax for `static` and make it a `validate` error to place `static` on component or primitive definitions themselves.
closed 2022-07-29 by rachitnigam