calyxir/calyx #1104: Non-Xilinx AXI test harness

sampsyo, 2022-07-15:
Currently, we have two ways of running Calyx-generated RTL:

* Simple hardware, open-source simulator: This is the "normal" way of running Calyx programs. We compile the core design to Verilog and use Verilator or Icarus Verilog to run them with a very minimal test harness that just consists of "magically" loading in the contents of memories via Verilog's `readmemh`.
* AXI-wrapped hardware, Xilinx simulator: This is what our Xilinx toolchain stuff does. We compile the design with a big complicated AXI wrapper, and we simulate it with our PyOpenCL or PYNQ harness and the [XRT][] emulation mode (which uses xsim underneath). This is important because it's closer to "real" hardware execution, but it's inconvenient and slow because it has to use Xilinx stuff.

These two modes couple two dimensions:

* Is the hardware simple (just the core design) or AXI-wrapped?
* Do we use open-source simulators or Xilinx tools?

It would be really nice to decouple these choices, creating a third mode where we run *AXI-wrapped* hardware on *open-source simulators*. This would make it much faster and less painful to debug the AXI wrapper itself—as conveniently as we currently debug just the "core" hardware.

This would entail, as discussed in https://github.com/cucapra/calyx/discussions/1022#discussioncomment-2965475 and thereafter, writing a testbench that implements the "host side" of the AXI protocol for a given design. There are several options on the table for doing that:

* Build on @sgpthomas's [existing implementation for Verilator](https://github.com/sgpthomas/axi-playground).
* Use cocotb's existing [AXI test harness](https://github.com/alexforencich/cocotbext-axi).
* Write the AXI behavior in non-synthesizable Verilog, along the lines of [this blog post](https://klickverbot.at/blog/2016/01/testing-verilog-axi4-lite-peripherals/). Some [existing open-source Verilog modules](https://github.com/pulp-platform/axi#simulation-only-modules) might help with this.

[xrt]: https://github.com/xilinx/xrt

rachitnigam, 2022-07-18:
This would be great and is something we should prioritize doing before the end of the summer!

andrewb1999, 2022-07-19:
FYI, this exists: https://github.com/alexforencich/cocotbext-axi. I've used it before with some success but it can be a little challenging to get working. My main suggestion here is to connect the m_axi ports to pre-existing axiram IPs that are initialized with the correct test data and only use cocotbext-axi for communicating with the axilite control.

nathanielnrn, 2022-07-21:
After discussing with @rachitnigam, the goal is to write a [cocotb](https://github.com/cocotb/cocotb) [AXI test harness](https://github.com/alexforencich/cocotbext-axi) that would use Icarus Verilog (Verilator is only supported on version 4.106).

nathanielnrn, 2023-01-01:
Bumping this because this might be able to be closed as runt now contains tests that emulates AXI-wrapped hardware using cocotb on our vectorized-add and dot-product examples.

The runt tests are here:
https://github.com/cucapra/calyx/blob/c6f750af5793e3e090d901e48194c5156884d787/runt.toml#L291-L310

sampsyo, 2023-01-03:
Awesome!! Yep, I think we are done here. This is incredibly useful to have!

closed 2023-01-03 by sampsyo