calyxir/calyx #1127: vcd.dump files not created on passing -s fpga.waveform true flag

nathanielnrn, 2022-07-25:
Running fud as follows:
`fud e vad-emulation.xclbin --from xclbin --to fpga -s fpga.data .../vectorized-add.fuse.data -s fpga.waveform true -s fpga.save_temps true` 
Does not produce a dump.vcd file in `.run/*/hw_em/device0/binary_0/behav_waveform/xsim/` as expected.

nathanielnrn, 2022-07-31:
So it turns out the key was [assigning the environment variable](https://github.com/cucapra/calyx/blob/7cb5cf010f32dbe43a64d17b5c70bab964d87821/fud/fud/stages/xilinx/execution.py#L83) `XRT_INI_PATH` only __after__ the `xrt.ini` file is [written](https://github.com/cucapra/calyx/blob/7cb5cf010f32dbe43a64d17b5c70bab964d87821/fud/fud/stages/xilinx/execution.py#L50-L65).
At least, thats what seemed to be the difference, not going to question it.

rachitnigam, 2022-08-01:
We should probably leave a comment in the source code saying that this needs to be the case

closed 2022-08-03 by nathanielnrn