calyxir/calyx #1131: Case that infer-share does not detect

calebmkim, 2022-07-29:
Currently, `infer-share` does not detect `foo` as shareable. 
```
component foo() -> () {
  cells {
    lt = std_lt(32);
    r = std_reg(32);
    w = std_reg(32);
  }
  wires {
    comb group cmp {
      lt.left = 32'd2; 
      lt.right = 32'd4;
    }
    group wr_1 {
      r.write_en = 1'd1;
      r.in = 32'd0;
      wr_1[done] = r.done;
    }
    group wr_2 {
      r.write_en = 1'd1;
      r.in = 32'd0;
      wr_2[done] = r.done;
    }
    group read {
      w.in = r.out;
      w.write_en = 1'd1;
      read[done] = w.done;
    }
  }
  control {
    seq {
      if lt.out with cmp {
        wr_1;
      } else {
        wr_2;
      }
      read;
    }
  }
}
```
Even though we can see  that `r` will be written to before being read from, since we know at least one of the if branches will execute, we can't say whether there is a _specific_  write to `r` that dominates the read from `r`, and therefore DominationMap does not detect the read from r in `read` as being dominated by a write.

rachitnigam, 2022-11-13:
Oh, this is interesting! @sampsyo any thoughts on what kind of analysis would be useful here? Domination is about specific writes but we almost want an "approximate" domination analysis that only checks if there is always *some* write to the variable that dominates the read

sampsyo, 2022-11-14:
Interesting indeed!

Is it too simplistic to suggest that the data-flow analysis we need would ask: "are there live ranges that reach the top of the program"?

That is, the thing that makes a component non-shareable is when might read `r` before writing to it. Thinking about this in data-flow terms, that kind of thinking views reads creating live ranges that "face upward" in the CFG; writes "close off" these ranges and stop their propagation. A variable that is live all the way at the top is a problem.

A merge function in such an analysis could conclude that `r` is guaranteed to be overwritten on both branches.

rachitnigam, 2023-04-29:
Turning this into "Needs Triage" because it's clear that we should use some sort of live range analysis to answer this question but we need to figure out exact details. @calebmkim if it seems obvious to you then maybe you can write a short paragraph here and add the "available" label instead?