calyxir/calyx #1133: Interpreter has difficulty with signed numbers

susan-garry, 2022-07-30:
When I run a simple example with negative numbers through a 2d-matrix multiplier, I get two different outputs depending on if I use verilator or the interpreter (although the result of the multiplication is the same).

This is the example:

```
{
  "out": {
    "data": [[0, 0], [0, 0]],
    "format": {
      "numeric_type": "bitnum",
      "is_signed": true,
      "width": 32
    }
  },

  "A": {
    "data": [[1, 0], [-1, 1]],
    "format": {
      "numeric_type": "bitnum",
      "is_signed": true,
      "width": 32
    }
  },

  "B": {
    "data": [[1, 0], [1, 1]],
    "format": {
      "numeric_type": "bitnum",
      "is_signed": true,
      "width": 32
    }
  }
}
```

These are the final values in memory when I run it through verilator:

```
"memories": {
    "A": [
      [
        1,
        0
      ],
      [
        -1,
        1
      ]
    ],
    "B": [
      [
        1,
        0
      ],
      [
        1,
        1
      ]
    ],
    "out": [
      [
        1,
        0
      ],
      [
        0,
        1
      ]
    ]
  }
```

And these are the final values when I run it through the interpreter:

```
"memories": {
    "main": {
      "A": [
        [
          1,
          0
        ],
        [
          4294967295,
          1
        ]
      ],
      "B": [
        [
          1,
          0
        ],
        [
          1,
          1
        ]
      ],
      "out": [
        [
          1,
          0
        ],
        [
          0,
          1
        ]
      ]
    }
  }
```

Since `is_signed` is true for all arrays, it seems like the interpreter may not be handling signed numbers.

EclecticGriffin, 2022-07-30:
If you don't pass the `--raw` flag, the interpreter will just output an unsigned int interpretation of all numbers. With that flag fud should do some fun data formatting. See https://github.com/cucapra/calyx/blob/master/fud/fud/stages/interpreter.py#L162

There are also some examples in the runt test suite for the interpreter.

I should perhaps add that flag to the default config for fud. 

Anyway, if running with that flag doesn't fix things then there is likely an error in fud's post-interpretation data processing code

sampsyo, 2022-07-31:
Interesting point! Yeah, maybe that should indeed be the default… if for no other reason than "Verilator and interpreter modes match" is a pretty desirable property.

susan-garry, 2022-08-02:
Hmm. When I try something like `fud e --raw --to interpreter-out -s verilog.data calyx-tutorial/2d-matmul1.json calyx-tutorial/2d-matmul.futil`, I get this error message: `unrecognized arguments: --raw`. I also can't find any documentation for this option. Where might I find an example of how to use the `--raw` flag?

sampsyo, 2022-08-02:
Good question—in fact, `--raw` is an argument to the actual interpreter executable. So if you were to run that directly (not through fud), you would use `--raw`. To pass this through fud, you'll need to use [the `interpreter.flags` config option](https://github.com/cucapra/calyx/blob/8bf819963f212c3cd8ef70657af386e2f36e9119/fud/fud/config.py#L35). So something like `fud e -s interpreter.flags '--raw' ...` or similar.

EclecticGriffin, 2022-08-25:
@susan-garry, is this issue still happening? If so, could you send me the program & data so I can investigate

closed 2022-09-03 by EclecticGriffin