calyxir/calyx #1135: Live Range Analysis dealing with While Loops

calebmkim, 2022-08-02:
Currently LiveRangeAnalysis incorrectly shares `r1` and `r2` in the following program: 
```
component main() -> () {
  cells {
    r1 = std_reg(32);
    r2 = std_reg(32);
    x = std_reg(32);
  }
  wires {
    group wr_r1{
      r1.write_en = 1'd1;
      r1.in = 32'd0; 
      wr_r1[done] = r1.done; 
    }
    group wr_r2{
      r2.write_en = 1'd1;
      r2.in = 32'd7; 
      wr_r2[done] = r2.done; 
    }
    group rd_r2{
      x.write_en = 1'd1; 
      x.in = r2.out; 
      rd_r2[done] = x.done; 
    }
  }
  control {
    seq {
      wr_r1; 
      wr_r2;
      rd_r2; 
      while r1.out {
        wr_r1; 
      }
    }
  }
}
```
This is because of the code to handle the `while` pattern match: 
```
ir::Control::While(ir::While { body, port, .. }) => {
            // recursive call on the body of the while loop 
            let (mut alive, gens, kills) =
                build_live_ranges(body, alive, gens, kills, lr);
            // adding the port name to the live cells 
            if let Some(cell) =
                LiveRangeAnalysis::port_to_cell_name(port, &lr.state_share)
            {
                alive.insert(cell)
            }
            // another recursive call on the body of the while loop (with the possibly updated alive set)
            return build_live_ranges(body, alive, gens, kills, lr)
```
In other words, the "flow" of the analysis is: body -> port -> body. So the last thing the live range analysis analyzes is the body of the while loop. In the given example, the the value in `r1` os killed in the body of the while loop, which means that the value in `r1` is not live as it exits the while loop and enters `rd_r2`. (Not sure if this makes sense or not...) 
It seems like the "flow" should actually be: port -> body -> port, since the body is always directly preceded and succeeded by a read of the port.
Changing the code to the following fixes the problem. 
```
ir::Control::While(ir::While { body, port, .. }) => {
             if let Some(cell) =
                LiveRangeAnalysis::port_to_cell_name(port, &lr.state_share)
            {
                alive.insert(cell)
            }
            let (mut alive, gens, kills) =
                build_live_ranges(body, alive, gens, kills, lr);
            if let Some(cell) =
                LiveRangeAnalysis::port_to_cell_name(port, &lr.state_share)
            {
                alive.insert(cell)
            }
            (alive, gens, kills)
```
I'm not sure if this is completely correct, though.

calebmkim, 2022-08-02:
Another comment: I was thinking of trying to get `LiveRangeAnalysis` to handle comb groups so that is why this was on my mind.

rachitnigam, 2022-08-03:
Sounds good. Can you add a test when you fix this?

PR #1141 (2022-08-05): Fixes LRA bug (#1135)
Also added a test case for this situation. Fixes #1135
closed 2022-08-05 by calebmkim