calyxir/calyx #1136: `group2invoke` disables a bunch of passes

rachitnigam, 2022-08-02:
The `MergeStaticPar` and `StaticParConv` only act upon groups and cannot be run on `invoke` statements. Since `group2invoke` turns a lot of simple groups into `invoke` statements, these passes are less effective. We should probably redesign the pass pipeline so that these passes run after `compile-invoke`

calebmkim, 2022-08-04:
Once I get LiveRangeAnalysis to Handle Comb Groups, here's what I was thinking: 

```
GroupToSeq,
GroupToInvoke, // Creates Dead Groups potentially
ComponentInliner,
CombProp,
CompileRef, //Must run before cell-share.
InferShare,
CellShare, // LiveRangeAnalaysis must handle comb groups
RemoveCombGroups, // Must run before infer-static-timing
InferStaticTiming,
CompileInvoke
MergeStaticPar,
StaticParConv,    // Must be before collapse-control
DeadGroupRemoval, // Since MergeStaticPar and GroupToInvoke potentially create dead groups
CollapseControl,
```

rachitnigam, 2022-08-05:
Does LRA handle comb groups? Also, if we can move somehow move `compile-invoke` to before `remove-comb-groups`, we can get rid of the extra cycle cost we have to pay for the `invoke-with` syntax

calebmkim, 2022-08-05:
1) LRA doesn't handle comb groups yet, although I can start working on that-- we can discuss more synchronously, but I've been thinking about it like this: 
for `invoke _ with comb group` we can just set all components used in `comb group` as live in the invoke. 
for `if _ with comb group` or `while _ with comb group` we can analyze the live set at the comb group itself. In other words, comb groups that go w/ if and while statements should be treated as their own "nodes" in the LiveRangeAnalysis.

2) I think just moving `compile-invoke` to be right before `remove-comb-groups` should work, unless there's some problem with that that I'm missing. I wasn't thinking about the timing we would save from that.

rachitnigam, 2022-08-08:
Currently infer-static-timing benefits from having invokes since it can simply copy the latency of the component to the static time of the invoke. If we do compile-invoke beforehand, we lost thing but it might be a worthwhile trade-off

rachitnigam, 2022-09-28:
@calebmkim am I misremember or did you open a PR to fix this during the summer?

calebmkim, 2022-09-29:
So https://github.com/cucapra/calyx/pull/1144 handles this particular issue, in the sense that invokes are compiled into groups so that the `MergeStaticPar` and `StaticParConv` sees groups. However, there may be some work to do in terms of improving the ordering of the passes even more.

rachitnigam, 2022-09-30:
Okay, let's close this particular issue in that case

closed 2022-09-30 by rachitnigam