calyxir/calyx #1146: Calyx Sharing Results-- To-do Tasks

calebmkim, 2022-08-15:
Main tasks 

- [ ] Start generating more kernels through TVM 
Need to decide a) which other kernels to use besides lenet (probably VGG 16 and SqueezeNet but we need others as well) and b) update the python tvm-to-calyx files to support the new TVM operations that using these new kernels will need. 

- [ ] Get full resource numbers from vivado 
Instead of just getting the resource estimates from the synthesis files, want to get resource estimates from the implementation files too (and within a reasonable amount of time). Key is to map primtive memories to UltraRAM instead of BRAM. 

- [ ] Actually Run the Experiments 
Run each .futil files on using: 1) no sharing, 2) always sharing, and an 3) ablation study by varying 1 bound while keeping the other two constant. Then we can see results.

rachitnigam, 2022-08-22:
Woo, awesome! Thanks for getting this started. Pondering what minimal version of #1151 is needed to get reasonable results for sharing

rachitnigam, 2022-11-13:
@calebmkim can you update this tracker to reflect the current state of things. If we've moved the tracking to the evaluation repository, let's close this issue with a link to that tracker/project

calebmkim, 2022-11-13:
There's also the project I started a couple weeks ago: it's basically a checklist (with a link to the PRs/Issues) of the things I've done to get Calyx Sharing on Big Designs. 

So basically rn everythingis kind of scattered: I'd be happy to just update this issue to reflect the progress I've made. Or do you think it would make more sense for me to start a new discussion in the sharing repo?

sampsyo, 2022-11-14:
I really like Projects for this kind of thing, IMO! But @calebmkim, it's really about whatever will make you feel most organized and efficient. :)

rachitnigam, 2022-11-14:
Agreed with @sampsyo

calebmkim, 2023-04-29:
(ended up using [project board](https://github.com/cucapra/calyx/projects/6))

closed 2023-04-29 by calebmkim