calyxir/calyx #1180: Live Range Analysis Not Sharing Multipliers

calebmkim, 2022-09-17:
Consider the following code, where `mult1` and `mult2` are multipliers: 
```
* mult1 and mult2 are live here * 
par{
    seq{write_mult1; read_mult1;}
    seq{write_mult2; read_mult2;}
}
```
Currently Calyx marks `mult1` and `mult2` as live on the output of the par block (i.e., above the par block). 
The reason why is that it looks at the reads of `mult1` and `mult2` and adds these cells to the `gens` of each seq block. Then, it takes: `(U(alive) - U(kills) + U(gens)) ` as the live set on the output of the par block, meaning that `mult1` and `mult2` are live. The problem is that, while it is true that  `mult1` and `mult2` are indeed read from, and therefore part of gens, in the par block, they are also guaranteed to be killed before we get to the output (i.e. beginning) of the par block. Currently Calyx does not consider the relative ordering of gens and kills in par 

A very easy workaround I came up with is to do this. We keep the same equation -- `(U(alive) - U(kills) + U(gens))` -- except we only count the gens that are alive on the output of at least one thread in the par block. Notice that this would solve the problem presented above, as both `mult1` and `mult2` are not alive on the output of their respective seq blocks.

rachitnigam, 2022-09-17:
I think this passes the smell test. However, I would like to visualize this a little better before committing to this–dataflow equations are nefariously complicated. Can you contrast this to the behavior in `if` blocks and maybe draw some diagrams to explain what's going on here?

calebmkim, 2022-09-19:
So the main issue I'm concerned with is the `gen` and `kill` of par blocks. First, here is my understanding of what `gen` and `kill` mean. `gen(A)` is any cell that is read from at any point in the control node A, while `kill(A)` is any cell written to at any point in point in the control node A.

I've attached a drawing of the current implementation. 

[current-par-liveness.pdf](https://github.com/cucapra/calyx/files/9600606/current-par-liveness.pdf)

This is fine since it conservatively overestimates the live ranges. But my idea was that we can do better than this. 

Specifically, we can relax constraint b) from my drawing. It's not that the cell can _never_ be read from in any of the children/threads fo the par block. It's that, if the cell is read from, we have to be sure that the read of the cell is dominated by a write to the cell that is within the same thread. 

So, how do we know that each read of a given cell must be dominated by a write within the same thread? 
For each cell in `U gen(children)`, we can just check if that cell is alive on the output of any child. If it isn't then we know every read of that cell must have been dominated by a write within the same thread (otherwise it would have been alive at the output of that thread). There may be other ways to answer this question, though. 

I should also note that even though I say "every read of that cell must have been dominated by a write within the same thread", we know this will only be the case when the cell is read from in a single thread. If a cell is read from in multiple different threads, then it is impossible for both threads to include a write that dominates the reads, since we are dealing with par blocks, and therefore, we must conclude that the cell is live on the output of the par block.

calebmkim, 2022-09-21:
Ok, here is a summary of the current ideas we have about par liveness (This is really long, but I wanted to get a comprehensive wrap-up + summary about what we talked about in the Calyx meeting yesterday). 

**Terminology I am using*

First, since CFGs are "backwards", to clear up terminology, when I say the input/output of a par block, I am thinking in terms of CFG. So the "input" of the par block occurs at the end of the par block (in terms of regular control flow) and the "output" of a par block is really the beginning of the par block. In other words, I say "input/output" to refer to my thinking about CFGs and "beginning/end " to refer to my thinking about regular control flow. 

[image]

**Explanation**

[Most of this was discussed in the Calyx meeting yesterday] 

**Current Problem (Discussed in Calyx Meeting)**

Currently, the Calyx [documentation](https://docs.calyxir.org/optimizations/dataflow.html) explains why simply taking `U alive(children)` or `∩ alive(children)` will not work.  Instead, it uses: `alive(par) = (U alive(children)  -  U kill(children)) + U gen(children)`. (and to be clear, `U alive(children)` means " the union of everything alive on the output of each child") 

Currently, `gen` and `kill` are all the reads and writes respectively that occur in a given child. Notice that this calculation of gens and kills is  _not sensitive to the ordering in any way_. This can lead to an overestimation of the live range. Consider the following example. 

[image]

Here, `x` would be in `gen(B)` and therefore alive on the output of the par block, even though we can clearly see that it is not, since the write to `x` in `B` must have killed `x`. 

**My Solution (discussed in Calyx Meeting)**

Therefore, I proposed a new idea. This rethinks the way we calculate `gen` to be more order-sensitive. I'll refer to it as `newgen`. 

For any group enable or invoke `G`: 
`newgen`: `output(G) = (input(G) - writes(G)) + reads(G)`  instead of 
`gen`: `output(G) = input(G) + reads(G)`  

To be clear, if I say cell `x \in newgens(C)` for some control node C, I mean it is part of the set `newgens` on the output of C. 

Notice that undert this equation, `x` would not be a not part of `newgen` on the output of `B`. 

At first, this seems weird: isn't `newgen` calculated in the exact same way as `alive` is? The key difference is that, whenever we come across a par block and do dataflow analysis, `gen` and `kill` are initialized to be empty at the input of each child of the par block. This is very important. Consider the following 2 examples. 

[image]

The live ranges on the output of each child are identical in both situations. However, the `newgen(A)` is different. 

In the the first example, since `newgen` starts out empty on the input to each child, `x` is never added to `newgen` and therefore not in `newgen(A)`.  Therefore when we do `alive(par) = (U alive(children)  -  U kill(children)) + U newgen(children)`, `x` is not added as live on the output of the par block, which is the behavior we want. 

In the second example,  `x` is added to `newgen(A)` then it is read from in `A`.  Therefore when we do `alive(par) = (U alive(children)  -  U kill(children)) + U newgen(children)`, `x`  is added as live on the output of the par block, which is the behavior we want. 

As you can see, `newgen` allows us to ask which cells are alive on the output of a given thread _because of some read that occurred within that thread_  (versus already being alive on the input to the thread). 

**Thinking about my solution more closely and addressing possible concerns (not discussed in Calyx meeting)**

There are a couple other concerns you might have as well (please feel free to respond if you think there is another concern I am not addressing here). 

1) Does this `newgen` calculation work with `if` blocks? 

We calulcate `newgen(output) = newgen(A) + newgen(B) + (any reads that occur in the if conditional)`. Therefore, the only time `newgen` differs from `gen` (the old calculation) is the following situation: when both `A` and `B` kill some cell `x` and that cell is not read from later on in the CFG. Notice that it is safe to exclude `x` from `newgen` in this situation, since it is guaranteed to be killed, no matter which branch we choose. If `x` were only killed in `B` for example, but was read from (without a kill) in `A`, then `newgen` would include `x` on the output of `A` and therefore on the output of the `if` statement when we take the union. Here is a sketch: 

[image]

2) Don't kills have to be order sensitive too in this case? Shouldn't we have something like `newkills: output(G) = inpupt(G) + writes(G) - reads(G)`? 

This isn't necessary, because any `gens` always take precedence when determining liveness over `kills`. The only time it changes the composition of `kills` is when some cell `x` is removed from `kills` since it was read from in some group/invoke. Then `x` will also be added to `newgens`. If `x` remains in `newgens` throughout the rest of the CFG, then even though we take away `x` when we do `-  U kill(children)` we are going to add it back when we do `+ U newgen(children)`, so either way `x` will end up as live on the output of the par block. If `x` is removed from `newgen` at any time throughout the rest of the CFG, that means that it must have be added back to `kill` at that time, meaning it was fine for us to keep `x` in kills. So you can see how using the proposed equation doesn't lead to any difference in the calculation of any live ranges. 

3) What about while loops? Consider the following diagram. 

[image]

We can see that, even within the while loop, it is safe to eliminate as `x` as part of the newgens on the output of the par block. However, because of the conditional, the body of the while loop that occurs (in terms of control flow) after the end of the par block, cannot share `x`.

rachitnigam, 2022-09-26:
Okay, thanks for clarifying it! I think the idea makes sense and it seems a stricter live-range computation than before which means olders programs should continue working

PR #1181 (2022-09-28): Update Live Range Analysis 
Resolves #1180.
closed 2022-09-28 by calebmkim