calyxir/calyx #1194: Recommend Docker image in getting started instructions

rachitnigam, 2022-09-30:
Our CI now uses the docker image. We should make it the recommended way to work with Calyx. Couple of complexities I ran into when setting up CI that we should talk about in the instructions:

1. `PYTHONENV` needs to contain `flit`'s install location. See the Dockerfile's last few lines to get the exact `ENV` command needed.
2. `fud check` might fail if the user running the image is not `root`. Recommend people copy `/root/.config/*` into `~/.config/` so that they get the `fud` configuration

rachitnigam, 2022-10-03:
Note that the Calyx and Dahlia repos need to be updated when using the containers

anshumanmohan, 2023-05-18:
Requesting that the Docker start off with `turnt` installed! Requesting because #1482 adds testing for MrXL.

PR #1494 (2023-05-21): Recommend using docker container in tutorial
Fixes #1194.
closed 2023-05-21 by rachitnigam