calyxir/calyx #1200: Unused port removal

rachitnigam, 2022-10-05:
Add a pass that eliminates unused ports of a component. In doing this, we also get rid of any assignments to those ports and allow the various dead-cell elimination passes to work better.

## TODO
- [ ] Add a new attribute called `@internal` or something to state that a component's signature does not need to be maintained to the outside world.
- [ ] Implement the pass to remove the unused ports

sampsyo, 2022-10-12:
Super dumb question: is this a whole-program analysis? That is, does it need to make a closed-world assumption—that we know all the instantiations of every component, so we can tell that a given port is never used via any of them?

rachitnigam, 2022-10-12:
Yup, definitely a whole program analysis

sampsyo, 2022-10-12:
Awesome; thanks for clarifying.