calyxir/calyx #1220: Cannot invoke the same component twice when using `ref` keyword

susan-garry, 2022-10-27:
This seems to be true in all circumstances. For example, `tests/correctness/ref-cells/ref.futil` works as is (with something like `fud e ref.futil --to data --through icarus-verilog -s verilog.data ref.futil.data` from within that directory). However, adding an extra `invoke adder[out = mem]()();` after line 64 gives the following error due to duplicate signal declarations:

``` 
=====STDERR=====
/tmp/tmpv4kndu6f:1397: error: 'adder_out_addr0' has already been declared in this scope.
/tmp/tmpv4kndu6f:1392:      : It was declared here as a variable.
/tmp/tmpv4kndu6f:1398: error: 'adder_out_write_data' has already been declared in this scope.
/tmp/tmpv4kndu6f:1393:      : It was declared here as a variable.
/tmp/tmpv4kndu6f:1399: error: 'adder_out_write_en' has already been declared in this scope.
/tmp/tmpv4kndu6f:1394:      : It was declared here as a variable.
/tmp/tmpv4kndu6f:1400: error: 'adder_out_read_data' has already been declared in this scope.
/tmp/tmpv4kndu6f:1395:      : It was declared here as a variable.
/tmp/tmpv4kndu6f:1401: error: 'adder_out_done' has already been declared in this scope.
/tmp/tmpv4kndu6f:1396:      : It was declared here as a variable.
```

Verilog gives a similar error message.

This isn't merely an issue of invoking the same component by passing in the same cell(s) twice; adding another memory component `mem1 = std_mem_d1(32, 1, 1)` and calling `invoke adder[out = mem1]()()` yeilds the same error. It also doesn't seem to be an issue with memories in particular, since `tests/passes/compile-ref/ref-invoke.futil` produces the same type of error, but passes registers by reference. Removing the second `invoke` statement from this file causes it to run without error.

rachitnigam, 2022-10-27:
@paili0628 thoughts on what could be going wrong?

rachitnigam, 2022-11-10:
Okay, did some digging. Seems just running `-p compile-refs` and storing the generated file does not cause the error anymore:
```
% cargo run -q -- tests/passes/compile-ref/ref.futil -p validate -p compile-ref -p validate > no-refs.futil
% cargo run -q -- no-refs.futil -p all -b verilog --disable-init --disable-verify > out.sv && verilator --lint-only out.sv
```
However, directly compiling the file with all passes generates an error:
```
% cargo run -q -- tests/passes/compile-ref/ref.futil -p all -b verilog --disable-init --disable-verify > out.sv && verilator --lint-only out.sv
%Error: out.sv:1397:11: Duplicate declaration of signal: 'adder_out_addr0'
 1397 |     logic adder_out_addr0;
      |           ^~~~~~~~~~~~~~~
        out.sv:1392:11: ... Location of original declaration
 1392 |     logic adder_out_addr0;
```

If it was a problem with `compile-ref`'s transformation then we would expect the two-step compilation to also produce the error message. However, it does not which means that the problem is hidden away in some data structure that is not being printed in the output.

paili0628, 2022-11-10:
Might it have something you do with the way we order our compiler passes in the default pipeline?

rachitnigam, 2022-11-11:
I was using a (fixed) version of the no-opt pipeline which only runs compilation passes and the error still exists so my guess would be no

rachitnigam, 2022-11-11:
More digging. When you change the code in `verilog.rs` using this patch:
```patch
diff --git a/src/backend/verilog.rs b/src/backend/verilog.rs
index 7d53e9dd..d3c6263e 100644
--- a/src/backend/verilog.rs
+++ b/src/backend/verilog.rs
@@ -264,7 +264,9 @@ fn emit_component(
 }
 
 fn wire_decls(cell: &ir::Cell) -> Vec<(String, u64, ir::Direction)> {
-    cell.ports
+    eprint!("{}: ", cell.name());
+    let ports = cell
+        .ports
         .iter()
         .filter_map(|port| match &port.borrow().parent {
             ir::PortParent::Cell(cell) => {
@@ -286,7 +288,9 @@ fn wire_decls(cell: &ir::Cell) -> Vec<(String, u64, ir::Direction)> {
             }
             ir::PortParent::Group(_) => unreachable!(),
         })
-        .collect()
+        .collect_vec();
+    eprintln!("{}", ports.iter().map(|(name, _, _)| name).join(", "));
+    ports
 }
 
 fn cell_instance(cell: &ir::Cell) -> Option<v::Instance> {
diff --git a/tests/passes/compile-ref/ref.futil b/tests/passes/compile-ref/ref.futil
index 91b8e289..145a96f2 100644
--- a/tests/passes/compile-ref/ref.futil
+++ b/tests/passes/compile-ref/ref.futil
@@ -61,6 +61,7 @@ component main() -> () {
 
   control {
     seq {
+      invoke adder[out = mem]()();
       invoke adder[out = mem]()();
       add_1;
       reg_to_mem;

```

Running:
```
% cargo run -q -- tests/passes/compile-ref/ref.futil -p no-opt -b verilog --disable-init --disable-verify > /dev/null
...
adder: adder_go, adder_clk, adder_reset, adder_done, adder_out_addr0, adder_out_write_data, adder_out_write_en, adder_out_read_data, adder_out_done, adder_out_addr0, adder_out_write_data, adder_out_write_en, adder_out_read_data, adder_out_done
```

The problem is somewhere (probably in `compile-ref`) we're adding ports to the `adder` cell for *each invocation* instead of just once creating this error

rachitnigam, 2022-11-11:
Ah, now that I look at `compile-ref`, this is obvious. We are adding new ports to the invoked cell every time we encounter it. The correct solution would be to go through all the cells once at the start, update their ports if we changed their signatures, and then just change the compilation for invoke to add only the port assignment.

@paili0628 does that make sense? It would also be good to make sure to understand how I debugged this problem. Feel free to ask questions

paili0628, 2022-11-11:
Yes it makes sense to me. I can change the compile-ref pass to only add ports at the start instead of adding it every time a cell is invoked

PR #1255 (2022-11-28): Ref problem
Fixes #1220
closed 2022-11-28 by paili0628