calyxir/calyx #1223: Vivado cannot synthesize VGG.

calebmkim, 2022-10-31:
Currently, when I try to get synth files for VGG, I get the following error: 

```
ERROR: [Synth 8-354] integer overflow when computing array size [/scratch/cmk265/learning/my-tmp/tmppz7msi7z/main.sv:29]
ERROR: [Synth 8-6156] failed synthesizing module 'seq_mem_d1' [/scratch/cmk265/learning/my-tmp/tmppz7msi7z/main.sv:8]
ERROR: [Synth 8-6156] failed synthesizing module 'seq_mem_d2' [/scratch/cmk265/learning/my-tmp/tmppz7msi7z/main.sv:95]
```

This has to do with the fact that VGG needs a 25088 x 4096 sequential 2d-memory in its design (which is really just a sequential 1-d memory withe size 102760448). This is less than 2^32-1, so I'm not exactly sure why it's saying "integer overflow". 

Either way, VGG uses a really big memory, and even if we do get it to work, it might take a really long time to get it working. I was thinking of trying to get googlenet to work. There are more operations, but the RAM required seems to be significantly less.  The main thing with googlenet is getting a reliable fixed point approximation, and I have a direction in mind to take.

rachitnigam, 2022-10-31:
That's pretty odd. Maybe worth investigating what happens if you run the flow again with a memory that's 1/10th the size and see if passes. The total size is probably 1MB or something so definitely on the larger size but might still fit on a URAM (would have to check the data sheet for the FPGA to figure that out)

sampsyo, 2022-10-31:
Just to be precise, 25088*4096 is about 100 million. If those are 32-bit numbers, that's 400 MB. So, definitely too big to fit in an on-chip memory, unfortunately.

rachitnigam, 2022-10-31:
Oops, Adrian is right! 400MB wouldn't fit on the device and if that layer cannot be changed somehow to use 1000x less memory, we're toast so focusing on smaller networks is definitely the way to go. I wonder if there are smaller versions of VGG that we could look at?

calebmkim, 2022-11-04:
One thought I had was this: the reason why we require such a large memory is really because of one operation, `dense`,  which does matrix multiplication with the transpose of an input tensor. Currently, we implement it by actually creating the transpose as a non-external memory (this is where the large memory problems come in), and then performs multiplication. 

I was thinking it may be possible to do this without actually instantiating the transpose as a non-external memory, and instead just do more work to get the matrix access indices correct.

rachitnigam, 2022-11-05:
Yeah, that’s definitely a possibility. There’s always these kinds of trade-offs you have to make: do you reify each array or figure out how to index it. Could be nice if the compiler had a switch to do that

sampsyo, 2022-11-05:
Yes—or even better, if TVM itself had a way to express those alternatives… for example, while Relay doesn't seem to have a "transpose-and-multiply" option AFACIT, the `topi` layer does include this as an option:
https://github.com/apache/tvm/blob/732e34fe3d9facf0d710970aed8c7a86c038b070/python/tvm/topi/transform.py#L512

(Namely, the signature for `matmul` includes Boolean flags to transpose either operand on-the-fly.) We are not working at the `topi` level, but it does make me wonder if there is some obvious way to do this more at the Relay level.

rachitnigam, 2022-11-15:
@calebmkim can `vgg` be synthesized with the recent changes you made? If so, let's close this

closed 2022-11-16 by calebmkim